projectacrn/acrn-hypervisor #5812: [documentation]

ionutnechita, 2021-03-03:
Hi ACRN Team,

I noticed in your documentation a small content problem.
From what I have now, the 2 NUC - NUC7i5BNH and NUC7i7BNH do not have a serial port, but I notice a model with a serial port.
https://projectacrn.github.io/latest/reference/hardware.html#hardware

Is it a documentation issue?

Kaby Lake(Code name: Baby Canyon) | NUC7i5BNH(Board: NUC7i5BNB)  --  Serial Port | Yes
And
Kaby Lake(Code name: Baby Canyon) | NUC7i7BNH(Board: NUC7i7BNB)  --  Serial Port | No

I know they both don't have a serial port.

Would that be correct?
Kaby Lake(Code name: Baby Canyon) | NUC7i5BNH(Board: NUC7i5BNB)  --  Serial Port | No
And
Kaby Lake(Code name: Baby Canyon) | NUC7i7BNH(Board: NUC7i7BNB)  --  Serial Port | No

gvancuts, 2021-03-03:
Good catch @ionutnechita !

There is an additional mistake in that doc, Dawson Canyon does have a serial port so we should also correct this line:

Kaby Lake (Code name: Dawson Canyon) | NUC7i5DNH (Board: NUC7i5DNB) -- Serial Port | No

Let me know if you were planning on sending a PR to fix this, else I'll do it a little later.

Thanks!

ionutnechita, 2021-03-03:
Hi @gvancuts,

If you could make the change in git for this issue. 
Thanks for getting involved.

gvancuts, 2021-03-03:
> If you could make the change in git for this issue.

PR #5813 sent for review

> Thanks for getting involved.

Thanks for bringing this up to our attention!

gvancuts, 2021-03-16:
Patch was merged, closing.

closed 2021-03-16 by gvancuts