projectacrn/acrn-hypervisor #5815: Doc:  Need update document to clarify kernel cmdline

NanlinXie, 2021-03-04:
Currenly we have two ways to config linux kernel's parameter
1. Definition in XML bootargs for pre-launched VM/Service VM
2. Grub.conf could overwrite XML bootargs via module2 line of grub.cfg.
  a. Appending to SOS via the multiboot2 command line (only for debug)  might be removed in the future.
3. Config-tool and HV code will change two kernel cmdline param (max cpu#/hugepagesz) due to the reason that wrong value will cause HV critical failure. Will document in future.

NanlinXie, 2021-04-12:
Regarding to "hugepagesz" and "hugepages" kernel cmdline, we are aligned to below solution:
1. Remove the hardcode setting in HV code.
2. User could use grub to overwrite these two settings.
3. Config tool will set default value based on cpuid support, hugepagesz=1G, and calculate hugepages.
4. Add parameter description for how to set these two parameters.
5. Add sanity check in config tool to check if hugepage parameter not supported in cpuid, eg: EHL.

szhen11, 2021-04-19:
[External_System_ID] ACRN-7057

dbkinder, 2022-01-26:
@NanlinXie @Kunhui-Li Can we close this issue now?