projectacrn/acrn-hypervisor #5818: [EHL][v2.4_rc1] Failed to boot up WAAG randomly

szhen11, 2021-03-05:
Environment
BOARD: EHL SBL
SOS Kernel: 5.4.90-linux-intel-ese-lts-acrn-sos-lts
Build Info: YOCTO-2021ww05.3 rc1
 
Steps
1. Boot to SOS
2. Boot up waag with script
 
acrn-dm -A -m $mem_size -s 0:0,hostbridge -U d2795438-25d6-11e8-864e-cb7a18b34643 \
--windows \
$logger_setting \
-s 6,virtio-blk,/root/waag/win10-ltsc.img \
-s 7,virtio-blk,/root/waag/newdisk.img \
-s 5,virtio-net,tap_WaaG,mac=00:16:3E:12:23:34 \
-s 2,passthru,0/2/0,gpu \
--ovmf /usr/share/acrn/bios/OVMF.fd \
$intr_storm_monitor \
-s 1:0,lpc \
-l com1,stdio \
-s 0:20:0,passthru,00/14/0,d3hot_reset \
$vm_name
 
 Expected result
boot up waag successfully
Actual result
sometimes failed to boot up waag , hang on the "welcom" screen , and cannot ping ip 
fail rate:2/10

szhen11, 2021-03-05:
[External_System_ID] ACRN-6984

ionutnechita, 2021-03-05:
Hi @gvancuts,

I also saw this issue today.
But with this change made:

-        <ENFORCE_TURNOFF_AC>n</ENFORCE_TURNOFF_AC>
+        <ENFORCE_TURNOFF_AC>y</ENFORCE_TURNOFF_AC>

This issue is not solved.
At startup/booting, virtual machine with Windows does not display an image on the screen.

szhen11, 2021-03-08:
@ionutnechita Could you describe the configs(board, scenario, launch script, WaaG version and graphics driver version) and steps and  when you see "At startup/booting, virtual machine with Windows does not display an image on the screen."?

closed 2022-03-04 by fuzhongl