projectacrn/acrn-hypervisor #5826: [WindowsGuest] Enable S5 - initiate a connection on a socket error

ionutnechita, 2021-03-08:
Hi ACRN Team,

I try to enable S5 support in WaaG, but observe this issue:

 ./win_vm1.sh
cpu1 online=0
cpu2 online=0
cpu3 online=0
ioctl(TUNSETIFF): Device or resource busy
device tap0 is already a member of a bridge; can't add it to bridge br0.
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF_debug.fd, size 0x200000
vm_create: win_vm1
VHM api version 1.0
vm_setup_memory: size=0x200000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:1/1 page size:0x200000
level 1 free/need pages:28/8 page size:0x40000000

try to setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x180000000
total_size 0x300000000

mmap ptr 0x0x7f570801e000 -> baseaddr 0x0x7f5740000000
mmap 0x80000000@0x7f5740000000
touch 2 pages with pagesz 0x40000000
mmap 0x180000000@0x7f5880000000
touch 6 pages with pagesz 0x40000000
mmap 0x200000@0x7f583fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x180000000
vm_init_vdevs
pm_by_vuart_init idx: 0, path: /run/acrn/life_mngr_win_vm1
initiate a connection on a socket error
create socket to connect life-cycle manager failed
Unable to init vdev (111)

 cat win_vm1.sh
#!/bin/bash
# Copyright (C) 2019 Intel Corporation.
# SPDX-License-Identifier: BSD-3-Clause

function launch_win()
{
vm_name=win_vm$1

#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

ip tuntap add tap0 mode tap
ip link set dev tap0 up
brctl addif br0 tap0
ip link set tap0 master br0

gpudevice=`cat /sys/bus/pci/devices/0000:00:02.0/device`

echo "8086 $gpudevice" > /sys/bus/pci/drivers/pci-stub/new_id
echo "0000:00:02.0" > /sys/bus/pci/devices/0000:00:02.0/driver/unbind
echo "0000:00:02.0" > /sys/bus/pci/drivers/pci-stub/bind

audiodevice=`cat /sys/bus/pci/devices/0000:00:1f.3/device`

echo "8086 $audiodevice" > /sys/bus/pci/drivers/pci-stub/new_id
echo "0000:00:1f.3" > /sys/bus/pci/devices/0000:00:1f.3/driver/unbind
echo "0000:00:1f.3" > /sys/bus/pci/drivers/pci-stub/bind

#usbdevice=`cat /sys/bus/pci/devices/0000:00:14.0/device`

#echo "8086 $usbdevice" > /sys/bus/pci/drivers/pci-stub/new_id
#echo "0000:00:14.0" > /sys/bus/pci/devices/0000:00:14.0/driver/unbind
#echo "0000:00:14.0" > /sys/bus/pci/drivers/pci-stub/bind

#for pm by vuart setting
pm_channel="--pm_notify_channel uart "
pm_by_vuart="--pm_by_vuart pty,/run/acrn/life_mngr_"$vm_name
pm_vuart_node=" -s 1:0,lpc -l com2,/run/acrn/life_mngr_"$vm_name

#for memsize setting
mem_size=8192M

acrn-dm -A -m $mem_size -s 0:0,hostbridge \
  -s 2,passthru,0/2/0,gpu \
  -s 3,virtio-blk,/usr/share/acrn/conf/win2k19-server.img \
  -s 4,virtio-net,tap0,mac=02:04:03:44:44:44 \
  -s 5,xhci,1-1,1-2 \
  -s 6,passthru,0/1f/3 \
  --ovmf /usr/share/acrn/bios/OVMF_debug.fd \
  $pm_channel $pm_by_vuart $pm_vuart_node \
  --windows \
  $vm_name
}

#  -s 5,xhci,1-1,1-2 \
#  -s 5,passthru,0/14/0 \

# offline SOS CPUs except BSP before launch UOS
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "$online" = "1" ]; do
                        sleep 1
                        echo 0 > $i/online
                        online=`cat $i/online`
                done
                echo $idx > /sys/class/vhm/acrn_vhm/offline_cpu
        fi
done

launch_win 1 "64 448 8"

Do you see any errors in the script?

SOS:
Acrn-dm  --version
HV / DM version is: 2.4-unstable-c843d2bbc-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-03-05 20:08:36
Uname -a
Linux demo-acrn-opensuse-lab1 5.4.90-PKT-200203T060100Z-00002-g2d8c7622646a #1 SMP PREEMPT Fri Feb 12 20:21:08 EET 2021 x86_64 x86_64 x86_64 GNU/Linux
OS: openSUSE Tumbleweed
Scenario: industrial / industry

fuzhongl, 2021-03-16:
@ionutnechita  Do you refer this [doc:](https://projectacrn.github.io/latest/tutorials/enable_s5.html) ?

ionutnechita, 2021-03-16:
Hi @fuzhongl,

Yes, this doc for S5 suport.

fuzhongl, 2021-03-16:
Do you launch other UOS before WaaG?
If yes, please share the launch script.
Thanks!

ionutnechita, 2021-03-16:
No, only Windows / WaaG.

fuzhongl, 2022-03-22:
@ionutnechita 
Seems like the same issue with #5936
Please refer the update doc:https://projectacrn.github.io/2.4/tutorials/enable_s5.html

```
Note
The User VM need to be authorized to be able to request a Shutdown, this is achieved by adding “--pm_notify_channel uart” in the launch script of that VM. And, there is only one VM in the system can be configured to request a shutdown. If there is a second User VM launched with “--pm_notify_channel uart”, ACRN will stop launching it and throw out below error message: initiate a connection on a socket error create socket to connect life-cycle manager failed
```

closed 2022-07-05 by fuzhongl