projectacrn/acrn-hypervisor #5844: [BaaG] BSD As A Guest - DragonFly not booting with ACRN Hypervisor

ionutnechita, 2021-03-16:
Hi ACRN Team,

I try to start DragonFly with ACRN-DM.
But when the kernel was initialized in BSD, it cracked with CPU panic in UOS / BaaG.

In SOS appear this messages:
[  236.721085] vcpu 0[1637]: segfault at 38 ip 0000557c6d159917 sp 00007fd123ffebf0 error 4 in acrn-dm[557c6d13b000+52000]
[  236.721091] Code: f8 02 0f 85 b1 00 00 00 4c 89 eb 48 c1 e3 04 4c 29 eb 48 c1 e3 03 4c 8d 7c 1d 60 4c 89 ff e8 c0 23 fe ff 4c 8b 74 1d 28 31 db <41> 8b 4e 38 85 c9 7e 28 90 49 8b 4e 08 48 63 c3 48 8b 04 c1 48 85

These BSD initialization messages appear on my display:
Fatal trap 9: general protection fault while in kernel mode

acrn-dm / acrn hv --version:
DM / HV version is: 2.4-unstable-c843d2bbc-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-03-05 20:08:36
acrn kernel version:
uname -a
Linux demo-acrn-opensuse-lab1 5.4.90-PKT-200203T060100Z-00002-g2d8c7622646a #1 SMP PREEMPT Fri Feb 12 20:21:08 EET 2021 x86_64 x86_64 x86_64 GNU/Linux

[dmesgPanicCPUBaaGInSOS.txt](https://github.com/projectacrn/acrn-hypervisor/files/6151371/dmesgPanicCPUBaaGInSOS.txt)
[launchBSD.txt](https://github.com/projectacrn/acrn-hypervisor/files/6151380/launchBSD.txt)
ISO: http://mirror-master.dragonflybsd.org/snapshots/x86_64/DragonFly-x86_64-LATEST-ISO.iso.bz2

gvancuts, 2021-03-23:
Hi @ionutnechita , am I understanding you correctly that you are trying to load the DragonFly BSD installer from a CD image and also giving ACRN an empty virtual disk, that I suppose you intent to use as the target installation disk?

What HW are you running on?

Have you tried to create a virtual disk with DragonFly properly installed on it using KVM/QEMU and then use that disk directly with ACRN (similar to what's described here: https://projectacrn.github.io/latest/tutorials/running_deb_as_user_vm.html)?

ionutnechita, 2021-03-23:
Hi @gvancuts,

// What HW are you running on?

I use this:
Kaby Lake
NUC7i7BNH/NUC7i7BNB