projectacrn/acrn-hypervisor #5853: /usr/share/acrn/conf doesn't exist!

ionutnechita, 2021-03-22:
Hi ACRN Team,

I noticed this:
/usr/share/acrn/conf doesn't exist!
_scan_added_vm: Failed to check directory /usr/share/acrn/conf, err: -1

Shouldn't the acrnd service do this folder? ( mkdir -p /usr/share/acrn/conf )
Or from installing hypervisor through makefiles.

acrn-dm --version
DM version is: 2.4-unstable-ae659e58a-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-03-22 12:36:01

gvancuts, 2021-03-22:
Not sure whether this is a fatal error or not (it does not look like one), it's been reported previously in this bug report: #4898. @ionutnechita , I propose that we close this one as a duplicate of this other bug report, any concern?

ionutnechita, 2021-03-22:
Yes, we can close it. To write in this ticket #4898 so we can fix it.

gvancuts, 2021-03-22:
Tracking the problem in #4898 , closing this.

closed 2021-03-22 by gvancuts