projectacrn/acrn-hypervisor #5854: /bin/sh: line 1: /usr/sbin/iasl: No such file or directory

ionutnechita, 2021-03-22:
Hi ACRN Team,

I noticed this:
/bin/sh: line 1: /usr/sbin/iasl: No such file or directory

SOS: 
OpenSUSE Tumbleweed

cat /etc/os-release
NAME="openSUSE Tumbleweed"
//# VERSION="20210320"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
//VERSION_ID="20210320"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20210320"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo"

//# which iasl
/usr/bin/iasl

OpenSUSE IASL is located in /usr/bin, but ACRN-DM he wants it in /usr/sbin.

acrn-dm --version
DM version is: 2.4-unstable-ae659e58a-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-03-22 12:36:01

...
cpu1 online=0
cpu2 online=0
cpu3 online=0
ioctl(TUNSETIFF): Device or resource busy
device tap0 is already a member of a bridge; can't add it to bridge br0.
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF_debug.fd.gop, size 0x200000
vm_create: win_vm1
VHM api version 1.0
vm_setup_memory: size=0x200000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:1/1 page size:0x200000
level 1 free/need pages:28/8 page size:0x40000000

try to setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x180000000
total_size 0x300000000

mmap ptr 0x0x7f7062cb5000 -> baseaddr 0x0x7f7080000000
mmap 0x80000000@0x7f7080000000
touch 2 pages with pagesz 0x40000000
mmap 0x180000000@0x7f71c0000000
touch 6 pages with pagesz 0x40000000
mmap 0x200000@0x7f717fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x180000000
vm_init_vdevs
Watching power button on /dev/input/event2
pci init hostbridge
pci init lpc
pci init passthru
polling 35...
Listening 35...
pci init virtio-blk
pci init virtio-net
pci init xhci
pci init passthru
pci init ahci
pci init ahci
pci init igd-lpc
tpm: init_vtpm2:Invalid socket path!
**_/bin/sh: line 1: /usr/sbin/iasl: No such file or directory_**
acpi_build failed, error=-1
pci deinit hostbridge
pci deinit lpc
pci deinit passthru
vm_reset_ptdev_intx:0x10-10, ioapic virpin=0.
pci deinit virtio-blk
pci deinit virtio-net
vtnet tx thread closing...
pci deinit xhci
pci deinit passthru
vm_reset_ptdev_intx:0x30-fb, ioapic virpin=0.
pci deinit ahci
pci deinit ahci
pci deinit igd-lpc
mngr_client_new: Failed to accept from fd 35, err: Invalid argument
Stop listening 35...
Stop polling 35...

gvancuts, 2021-03-22:
Hi @ionutnechita , yes, this is a known behaviour. Not one that I like but that's what it is :-)

This is why we have this paragraph in our "Getting Started" and "Building from Source" documents (note the last line):
```
$ wget https://acpica.org/sites/acpica/files/acpica-unix-20191018.tar.gz
$ tar zxvf acpica-unix-20191018.tar.gz
$ cd acpica-unix-20191018
$ make clean && make iasl
$ sudo cp ./generate/unix/bin/iasl /usr/sbin/
```

Be aware of this Pull Request too: https://github.com/projectacrn/acrn-hypervisor/pull/5849. You will need this since Tumbleweed has gcc-10 in it ;-)

ionutnechita, 2021-03-22:
Hi @gvancuts,

I understood this information. But the location can't change?
Must it be found every time in /usr/sbin? 
Can't change to /usr/bin?

In OpenSUSE, it can be installed from the official repository.
//# zypper in acpica
acpica-20210105-1.3.x86_64
Absolute path to 'iasl' is '/usr/bin/iasl'.

gvancuts, 2021-03-22:
You can change it at build time by setting the `ASL_COMPILER` variable, e.g.: `ASL_COMPILER=/usr/bin/iasl make BOARD=nuc7i7dnb SCENARIO=industry`.

gvancuts, 2021-03-22:
I was actually thinking we could easily make the build system locate the binary automatically so we don't have to set it (and we can then start using the distro `acpica` package instead of downloading and building manually)... here is a PR that does just that: #5856

gvancuts, 2021-03-25:
[External_System_ID] ACRN-7030

gvancuts, 2021-05-10:
I believe this can be closed now, please re-open it @ionutnechita if this is not the case.

closed 2021-05-10 by gvancuts