projectacrn/acrn-hypervisor #5859: Run Xenomai UOS with debian image failed

SIMING29, 2021-03-23:
Kabylake platform, acrn 2.3, Industry Scenario

Update the debian uos to kernel 4.19.59-rt24
The debian uos and Clear linux's xenomai uos worked well.
When I used the Debian img to replaced the Clear Linux img to run with Xenomai kernel, had the error "Kernel panic - not syncing: No working init found."

sudo ./launch_xenomai.sh -f ../vm-debian/debian10-40G.img -k ./linux-4.19.59-xenomai-3.1-acrn+-x86.tar.gz

`[    0.939639] Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance.
[    0.940055] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.19.59-xenomai-3.1-acrn+ #1
[    0.940274] I-pipe domain: Linux
[    0.940386] Call Trace:
[    0.940490]  dump_stack+0x98/0xb3
[    0.940617]  ? rest_init+0x60/0xb0
[    0.940726]  panic+0xdd/0x278
[    0.940863]  ? __do_execve_file.isra.44+0x9a/0x860
[    0.941048]  ? rest_init+0xb0/0xb0
[    0.941182]  kernel_init+0xe6/0x100
[    0.941319]  ret_from_fork+0x36/0x50
[    0.976364] Kernel Offset: disabled
[    0.976503] ---[ end Kernel panic - not syncing: No working init found.  Try passing init= option to kernel. See Linux Documentation/admin-guide/init.rst for guidance. ]---
`
[log.debian.xenomai.txt](https://github.com/projectacrn/acrn-hypervisor/files/6188071/log.debian.xenomai.txt)

gvancuts, 2021-03-23:
This sounds like perhaps it's not finding the right rootfs. Based on the log, it believes the rootfs is on /dev/vda6, which seems a bit odd and I suspect it's `/home` that's in there. What is the partition layout of your Debian image (`fdisk -l *.img`). I'm guessing the right rootfs partition is 2, can you try to force that on [this line](https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/config_tools/data/sample_launch_scripts/nuc/launch_xenomai.sh#L204)? I.e. `-B "root=/dev/vda2 rw...`

What is line 219 in your script? Is it `echo 3 > /sys/kernel/debug/dri/0/i915_cache_sharing`?

SIMING29, 2021-03-24:
@gvancuts  Thanks , it worked with change "root=/dev/vda2". I builded the Debian Image according to the latest tutorial, so the partiton is same like the[ Figure 48 ](https://projectacrn.github.io/latest/tutorials/running_deb_as_user_vm.html#partition-vda)

yes, the line 219 is "echo 3 > /sys/kernel/debug/dri/0/i915_cache_sharing"
and the file exists.
# ls /sys/kernel/debug/dri/0/i915_cache_sharing -al
-rw-r--r-- 1 root root 0 Mar  24 09:17 /sys/kernel/debug/dri/0/i915_cache_sharing

gvancuts, 2021-03-24:
> @gvancuts Thanks , it worked with change "root=/dev/vda2". I builded the Debian Image according to the latest tutorial, so the partiton is same like the[ Figure 48 ](https://projectacrn.github.io/latest/tutorials/running_deb_as_user_vm.html#partition-vda)

Thanks for the confirmation. We should really fix this [there](https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/config_tools/data/sample_launch_scripts/nuc/launch_xenomai.sh#L153), i.e. search for the right root partition instead of assuming it is the latest partition. I need to think a little about how to do it easily, I don't know that there is an easy way and if not perhaps we should add an argument so the user can specify it and only assume it's the last partition when that argument is not given. If you have thoughts (or even better a patch) to fix this, do not hesitate to share it ;-)

> yes, the line 219 is "echo 3 > /sys/kernel/debug/dri/0/i915_cache_sharing"
> and the file exists.
> ls /sys/kernel/debug/dri/0/i915_cache_sharing -al
> -rw-r--r-- 1 root root 0 Mar 24 09:17 /sys/kernel/debug/dri/0/i915_cache_sharing

I looked at that too and I believe it's because the underlying HW is Kaby Lake, and that's a gen9 GPU. Based on [this](https://github.com/projectacrn/acrn-kernel/blob/master/drivers/gpu/drm/i915/i915_debugfs.c#L3683-L3684), the driver is throwing a `-ENODEV` error when we're trying to write to that file.

gvancuts, 2021-03-29:
@NanlinXie who owns this script?

I think we should remove the `echo 3 > /sys/kernel/debug/dri/0/i915_cache_sharing` and it would be useful to have someone look at the logic for determining the rootfs as well.