projectacrn/acrn-hypervisor #5868: /dev/rtc0 to wait for clock tick timed out - synchronization failed

ionutnechita, 2021-03-24:
Hi ACRN Team,

I observe this issue:
demo-acrn-opensuse-labl1:/home/demo # hwclock --systohc
demo-acrn-opensuse-labl1:/home/demo # hwclock --hctosys
hwclock: select() to /dev/rtc0 to wait for clock tick timed out
demo-acrn-opensuse-labl1:/home/demo # hwclock --get
hwclock: select() to /dev/rtc0 to wait for clock tick timed out
demo-acrn-opensuse-labl1:/home/demo # hwclock --verbose
hwclock from util-linux 2.36.2
System Time: 1616596213.178276
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Last drift adjustment done at 1616596006 seconds after 1969
Last calibration done at 1616596006 seconds after 1969
Hardware clock is on UTC time
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
hwclock: select() to /dev/rtc0 to wait for clock tick timed out
...synchronization failed

acrn-dm --version
DM version is: 2.4-unstable-ae659e58a-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-03-22 12:36:01

uname -a
Linux demo-acrn-opensuse-labl1 5.4.90-PKT-200203T060100Z-00006-g07fb8047e3dc #1 SMP PREEMPT Mon Mar 22 13:00:34 EET 2021 x86_64 x86_64 x86_64 GNU/Linux

hwclock --version
hwclock from util-linux 2.36.2

Scenario:
Industrial / Industry

Branch:
master

fuzhongl, 2021-04-07:
@ionutnechita  please try `hwclock --directisa` 
Thanks!

ionutnechita, 2021-04-12:
Hi @fuzhongl,

This command working.
Thanks.

gvancuts, 2021-04-12:
Is this the expected behaviour (and if so, should this be documented somewhere) or is this a bug?

ionutnechita, 2021-04-12:
Hi @gvancuts,

Maybe it's a bug.
@fuzhongl, maybe you should explain this behavior here in the ticket so we can all understand it.

fuzhongl, 2021-04-12:
@JasonChenCJ 
Could you please help to comment about this issue?
Thanks!

closed 2022-07-05 by fuzhongl