projectacrn/acrn-hypervisor #5872: uart output "vpci_write_cfg 0:1c.1 not found! off: 0xdb, val: 0xbf" at acrn startup

SIMING29, 2021-03-25:
WhiskyLake platform,  acrn 2.3, Industry Scenario
When the acrn startup, the uart port continued to print the information as shown blow.
The SOS can start, I'm just not sure if this information has any effect on the system

`
[12866219us][cpu=0][vm0:vcpu0][sev=2][seq=37]:vpci_write_cfg 0:1c.1 not found! off: 0xdb, val: 0xbf

[12877043us][cpu=0][vm0:vcpu0][sev=2][seq=38]:vpci_write_cfg 0:1c.2 not found! off: 0xdb, val: 0xbf

[12887917us][cpu=0][vm0:vcpu0][sev=2][seq=39]:vpci_write_cfg 0:1c.3 not found! off: 0xdb, val: 0xbf

[12898796us][cpu=0][vm0:vcpu0][sev=2][seq=40]:vpci_write_cfg 0:1c.4 not found! off: 0xdb, val: 0xbf

[12909674us][cpu=0][vm0:vcpu0][sev=2][seq=41]:vpci_write_cfg 0:1c.5 not found! off: 0xdb, val: 0xbf
.......
......
`
[acrn-uart.log](https://github.com/projectacrn/acrn-hypervisor/files/6202423/acrn-uart.log)

SIMING29, 2021-03-25:
After the SOS started, when I launch a UOS , the ACRN hypervisor shell show the messages as below.
The UOS is works normally.
Could I ignore these error messages？

`
ACRN:\>
ACRN:\>[530151964us][cpu=0][vm0:vcpu0][sev=3][seq=236]:ret=-1 hypercall=0x80000080 failed in vmcall_vmexit_handler

[530163673us][cpu=0][vm0:vcpu0][sev=3][seq=237]:ret=-1 hypercall=0x80000080 failed in vmcall_vmexit_handler

[530175367us][cpu=0][vm0:vcpu0][sev=3][seq=238]:ret=-1 hypercall=0x80000080 failed in vmcall_vmexit_handler

[530187107us][cpu=0][vm0:vcpu0][sev=3][seq=239]:ret=-1 hypercall=0x80000080 failed in vmcall_vmexit_handler

[536457307us][cpu=0][vm1:vcpu0][sev=3][seq=242]:vlapic: Start Secondary VCPU1 for VM[1]...

`

zhangshuai75, 2021-04-23:
ignore these messages.

closed 2021-04-23 by zhangshuai75