projectacrn/acrn-hypervisor #5874: Build error when running multiple jobs in parallel

gvancuts, 2021-03-25:
The hypervisor build fails when running more than one job in parallel. Here is how to reproduce this:
```
$ make -j$(nproc) BOARD=nuc7i7dnb SCENARIO=industry RELEASE=n
```

The (truncated) output shows:
```
make -C /home/gvancuts/work/acrn-hypervisor/hypervisor BOARD=nuc7i7dnb SCENARIO=industry HV_OBJDIR=/home/gvancuts/work/acrn-hypervisor/build/hypervisor RELEASE=n
...
Board XML is fetched from /home/gvancuts/work/acrn-hypervisor/misc/config_tools/data/nuc7i7dnb/nuc7i7dnb.xml
Scneario XML is configuration fetched from /home/gvancuts/work/acrn-hypervisor/misc/config_tools/data/nuc7i7dnb/industry.xml
cc -c -g -O0 -std=gnu11 -D_GNU_SOURCE -m64 -Wall -ffunction-sections -Werror -O2 -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fno-strict-aliasing -fpie -Wall -Wextra -pedantic -I/home/gvancuts/work/acrn-hypervisor/misc/debug_tools/acrn_crashlog/include -I/home/gvancuts/work/acrn-hypervisor/misc/debug_tools/acrn_crashlog/include/public -fstack-protector-strong -I /home/gvancuts/work/acrn-hypervisor/misc/debug_tools/acrn_crashlog/common/include fsutils.c -o /home/gvancuts/work/acrn-hypervisor/build/misc/debug_tools/acrn-crashlog/common/obj/fsutils.o
/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/allocation.xml generated
/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/unified.xml generated
/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/config.mk generated
make[1]: *** No rule to make target '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/boards/board.c', needed by '/home/gvancuts/work/acrn-hypervisor/build/hypervisor/configs/boards/board.o'.  Stop.
make[1]: *** Waiting for unfinished jobs....
/home/gvancuts/work/acrn-hypervisor/build/hypervisor/include/config.h generated
make[1]: Leaving directory '/home/gvancuts/work/acrn-hypervisor/hypervisor'
make: *** [Makefile:98: hypervisor] Error 2
make: *** Waiting for unfinished jobs....
...
```

gvancuts, 2021-03-25:
I _think_ that simply adding the `hvdefconfig` target as a pre-requisite to the `hypervisor` target is the right fix. In my testing, it solved the problem all the time... I have just submitted PR #5875

gvancuts, 2021-03-25:
@junjiemao1 , can you please take a look?

junjiemao1, 2021-03-26:
My guess is that the root cause lies in some missing dependencies in the Makefile under `hypervisor/`. I'll take a closer look after understanding how parallel builds work with recursive invocations of `make`.

gvancuts, 2021-03-26:
> My guess is that the root cause lies in some missing dependencies in the Makefile under `hypervisor/`. I'll take a closer look after understanding how parallel builds work with recursive invocations of `make`.

That's my suspicion too, I don't know if there is a more subtle way of fixing this but take a look at PR #5875 as well. That one fixed the problem during all my testing.

gvancuts, 2021-03-26:
[External_System_ID] ACRN-7034

junjiemao1, 2021-03-29:
@gvancuts Here's my proposal: #5884.

gvancuts, 2021-03-29:
> @gvancuts Here's my proposal: #5884.

Thanks @junjiemao1 ! This looks good to me!

The patch I had proposed got merged just a couple of hours ago. Feel free to revert it if needed and merge this one instead!

gvancuts, 2022-01-25:
Patch merged, closing.

closed 2022-01-25 by gvancuts