projectacrn/acrn-hypervisor #5882: Nested virtualization in ACRN Hypervisor

ionutnechita, 2021-03-26:
Hi ACRN Team,

How could i activate Nested Virtualization in ACRN Hypervisor?
I have a Windows VM / WaaG machine and I would like to work with Hyper-V Manager and Docker containers.

[image]

In Turn Windown features on and off, Hyper-V Platform option cannot be enabled on.

How can i activate this?
Is this possible?

Scenario used:
Industrial / Industry

ACRN: 2.4

gvancuts, 2021-03-26:
Hi @ionutnechita , it is not possible to enable this, nested virtualization is not supported on ACRN.

ionutnechita, 2021-03-26:
Hi @gvancuts,

Do you know any possibilities how I can work with Docker containers in Windows?

gvancuts, 2021-03-26:
> Do you know any possibilities how I can work with Docker containers in Windows?

I don't I'm afraid.... I'm not an expert on Docker in Windows but my understanding was also that it required virtualization (Hyper-V) to be available... I don't know any way around that :-(

ionutnechita, 2021-04-12:
Hi @gvancuts,

I observed this ticket:
https://github.com/projectacrn/acrn-hypervisor/issues/5923

Maybe is possible in future. Maybe with ACRN 2.5.

Thanks.

gvancuts, 2021-04-12:
Hi @ionutnechita , yes, there is some form of nested virtualization that the team is looking into. I have not seen much details yet but my understanding is that the primary use-case is to allow this from within the Service VM. In your case, you would like to do this from a standard post-launched VM, and I'm not sure if that's within the scope of what they're looking at at the moment.

Having said that, @NanlinXie how can we capture this as a feature request to make sure the architects/engineers take it into consideration as part of #5923 ?

gvancuts, 2021-04-13:
@ionutnechita , FYI, the first set of patches to support nested virtualization have been sent on the mailing list for review: https://lists.projectacrn.org/g/acrn-dev/message/31213

closed 2022-07-05 by fuzhongl