projectacrn/acrn-hypervisor #5885: How to run Debian and Windows UOS at the same time

SIMING29, 2021-03-29:
Our goal is run a Windows and RTVM on the paltform. RTVM is best based on debian, not clr linux that we are not familiar with.
How to run the debian uos after the windows uos is running? 
The VGA controller and USB controller are all passthroughed to Windows UOS, so there is no display and usb device for the Debian.
I have tried launch them one by one, but always got "vm_create failed" error when launch the sceond uos.
`
[  759.664883] vhm_dev_open: opening device node
[  759.686358] vhm: failed to create VM from Hypervisor !
[  760.198191] vhm: failed to create VM from Hypervisor !
[  760.710005] vhm: failed to create VM from Hypervisor !
[  761.221819] vhm: failed to create VM from Hypervisor !
[  761.733637] vhm: failed to create VM from Hypervisor !
[  762.245553] vhm: failed to create VM from Hypervisor !
[  762.757375] vhm: failed to create VM from Hypervisor !
[  763.269050] vhm: failed to create VM from Hypervisor !
[  763.780736] vhm: failed to create VM from Hypervisor !
[  764.292404] vhm: failed to create VM from Hypervisor !
[  764.820636] Failed to destroy VM -1!
[  764.820643] vhm: freed vm

`

gvancuts, 2021-03-29:
Hi @SIMING29 , can you share the version of ACRN you are using, as well as the scenario and launch script?

I guess the Windows User VM is running correctly, is it?

SIMING29, 2021-03-30:
The paltform is Kabylake, ACRN V2.3, Industry Scenario

The uos works well when running Windows or debian respectively. 
The problem is that debian uos cannot be run after windows uos runs successfully.
[launch_debian.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/6227078/launch_debian.sh.txt)
[launch_win.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/6227080/launch_win.sh.txt)

gvancuts, 2021-03-30:
Hi @SIMING29 ,

There are a couple of things likely problematic in your launch script for Debian, can you try with the one below instead?

I have removed the following from it:
* `--windows` flag
* GPU sharing (aka GVT-g, `pci-gvt`) since you're giving the GPU to the Windows User VM as a pass-through device
* The Windows script gives the USB controller (PCI device 00:14.0) as a pass-through device to windows but there are a couple of USB busses/ports given to the Debian User VM. I have removed those.

```
#!/bin/bash
# Copyright (C) 2019 Intel Corporation.
# SPDX-License-Identifier: BSD-3-Clause

function launch_win()
{
vm_name=win_vm$1

#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

#for memsize setting
mem_size=4096M

acrn-dm -A -m $mem_size -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  -s 3,virtio-blk,./debian10-40G.img \
  -s 4,virtio-net,tap0 \
  --ovmf /usr/share/acrn/bios/OVMF.fd \
  $vm_name
}

# offline SOS CPUs except BSP before launch UOS
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
		# during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
		while [ "$online" = "1" ]; do
			sleep 1
			echo 0 > $i/online
			online=`cat $i/online`
		done
                echo $idx > /sys/class/vhm/acrn_vhm/offline_cpu
        fi
done

launch_win 1
```

gvancuts, 2021-03-30:
Actually... one more thing (which can potentially be very important). The script above for launching Debian is not optimized at all to run a Real-Time VM. I just realised though that this is what you want to do, so we should really look at a script that is more like the one below. A few words about that script (that you likely **cannot** use as is):
* Note the additional flags: `--rtvm`, `--lapic_pt`
* Note the UUID which the one for the 'hard real-time VM' (also defined [here](https://github.com/projectacrn/acrn-hypervisor/blob/v2.3/hypervisor/include/common/vm_uuids.h#L50-L52))
* We avoid sharing devices such as storage, ethernet hence the pass-through devices that you see there. That assumes you have at least two storage devices, i.e. one NVMe and one SATA disk
* That also assumes you have two network interfaces.

If you can send us the output of `lspci` and where root filesystems are stored, I can help you craft such launch script.

```
#!/bin/bash
# Copyright (C) 2019 Intel Corporation.
# SPDX-License-Identifier: BSD-3-Clause
# This is an example of launch script for KBL NUC7i7DNH, may need to revise for other platform.

# pci devices for passthru
declare -A passthru_vpid
declare -A passthru_bdf

passthru_vpid=(
["eth"]="8086 156f"
["sata"]="8086 9d03"
["nvme"]="8086 f1a6"
)
passthru_bdf=(
["eth"]="0000:00:1f.6"
["sata"]="0000:00:17.0"
["nvme"]="0000:02:00.0"
)

function launch_hard_rt_vm()
{
#for memsize setting
mem_size=1024M

modprobe pci_stub
# Ethernet pass-through
#echo ${passthru_vpid["eth"]} > /sys/bus/pci/drivers/pci-stub/new_id
#echo ${passthru_bdf["eth"]} > /sys/bus/pci/devices/${passthru_bdf["eth"]}/driver/unbind
#echo ${passthru_bdf["eth"]} > /sys/bus/pci/drivers/pci-stub/bind

# SATA pass-through
#echo ${passthru_vpid["sata"]} > /sys/bus/pci/drivers/pci-stub/new_id
#echo ${passthru_bdf["sata"]} > /sys/bus/pci/devices/${passthru_bdf["sata"]}/driver/unbind
#echo ${passthru_bdf["sata"]} > /sys/bus/pci/drivers/pci-stub/bind

# NVME pass-through
echo ${passthru_vpid["nvme"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["nvme"]} > /sys/bus/pci/devices/${passthru_bdf["nvme"]}/driver/unbind
echo ${passthru_bdf["nvme"]} > /sys/bus/pci/drivers/pci-stub/bind

# for pm setting
pm_channel="--pm_notify_channel uart "
pm_by_vuart="--pm_by_vuart tty,/dev/ttyS1"

/usr/bin/acrn-dm -A -m $mem_size -s 0:0,hostbridge \
   --lapic_pt \
   --rtvm \
   --virtio_poll 1000000 \
   -U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 \
   -s 2,passthru,02/0/0 \
   -s 3,virtio-console,@stdio:stdio_port \
   $pm_channel $pm_by_vuart \
   --ovmf /usr/share/acrn/bios/OVMF.fd \
   hard_rtvm

}

# -s 2,passthru,0/17/0 \ #please using "lspci -nn" check the bdf info

# Depends on which partation RT_LaaG is installed in;maybe need to change
# /dev/nvme0n1p3 to /dev/sda3 on NUC and uncomment SATA pass-through
# Add following RT_LaaG kernel cmdline into loader/entries/xxx.conf of EFI partation
#root=/dev/nvme0n1p3 rw rootwait nohpet console=hvc0 console=ttyS0 \
#no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 \
#clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 \
#intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 \
#isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll \
#irqaffinity=0

# offline SOS CPUs except BSP before launch UOS
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
                echo $idx > /sys/class/vhm/acrn_vhm/offline_cpu
        fi
done

launch_hard_rt_vm
```

SIMING29, 2021-03-31:
Hi @gvancuts Thanks for your detailed reply.

> There are a couple of things likely problematic in your launch script for Debian, can you try with the one below instead?

I edited the launch_debian.sh script and added a "-U 615db82a-e189-4b4f-8dbb-d321343e4ab3" parameter, the Debian vm is running according to the ACRN debug uart, but no system login displayed on the shell terminal.
Debian boot into blind mode. I think it's because the display device is passthroughed to Windows. 

```
ACRN:\>vm_list

VM_UUID                          VM_ID VM_NAME                          VM_STATE
================================ ===== ================================ ========
dbbbd4347a574216a12c2201f1ab0240   0   ACRN SOS VM                      Running 
d279543825d611e8864ecb7a18b34643   1   ACRN VM_1                        Running 
615db82ae1894b4f8dbbd321343e4ab3   3   ACRN VM_3                        Running 
3815882152084005b72a8a609e4190d0   4   ACRN VM_4                        Running 
ACRN:\>
ACRN:\>vm_console 3

----- Entering VM 3 Shell -----

Loading Linux 4.19.59-rt24 ...
Loading initial ramdisk ...
error: no suitable video mode found.
Booting in blind mode
```
VM_1 is Windows(works OK), VM_3 is Debian(works not OK), VM_4 is CLR Linux(works OK)

the shell termenal show a " not supported" massage
[log.launch_debian.txt](https://github.com/projectacrn/acrn-hypervisor/files/6235480/log.launch_debian.txt)
```
SW_LOAD: entry[2]: addr 0x000000003b800000, size 0x0000000004004000,  type 0x2
SW_LOAD: entry[3]: addr 0x0000000040080000, size 0x0000000000800000,  type 0x2
SW_LOAD: entry[4]: addr 0x0000000040880000, size 0x000000003f780000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000140000000, size 0x0000000080000000,  type 0x1
SW_LOAD: entry[7]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: entry[8]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu
                                     out instr on NMI port (0x61) not supported
                                                                               out instr on NMI port (0x61) not supported
                                                                                                                         out instr on NMI port (0x61) not supported
                                                                                                                                                                   Unhandled ps2 mouse command 0xe1
                        Unhandled ps2 mouse command 0x88

```

> * We avoid sharing devices such as storage, ethernet hence the pass-through devices that you see there. That assumes you have at least two storage devices, i.e. one NVMe and one SATA disk

Now I move to a WHL platform(ACRN V2.3, Industry Scenario), just have one disk, maybe cannot use launch_hard_rt_vm.sh.

I tried the launch_xenomai.sh, and have another issue.
sudo ./launch_xenomai.sh -f ./preempt-rt-32030.img -k ./linux-4.19.59-xenomai-3.1-acrn+-x86.tar.gz 
[log.launch_xenomai.txt](https://github.com/projectacrn/acrn-hypervisor/files/6235525/log.launch_xenomai.txt)
[launch_xenomai.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/6235528/launch_xenomai.sh.txt)

```
[ 1452.976107] vhm: failed to set irqline!
[ 1453.001445] vhm: failed to set irqline!
[ 1453.080370] vhm: failed to set irqline!
[ 1455.302539] vhm: failed to inject!
[ 1455.328906] vhm: failed to inject!
[ 1455.352640] vhm: failed to inject!
[ 1455.393550] vhm: failed to inject!

```

gvancuts, 2021-04-01:
> I edited the launch_debian.sh script and added a "-U 615db82a-e189-4b4f-8dbb-d321343e4ab3" parameter, 

Do  you mean you changed the UUID that was indicated in my launch script above? Can you share that latest script (that may be easier for me to look at).

> the Debian vm is running according to the ACRN debug uart, but no system login displayed on the shell terminal.
> Debian boot into blind mode. I think it's because the display device is passthroughed to Windows.

Most likely yes, there is no GPU available in that VM. That blind mode message is from Grub though as far as I know.

> ```
> ACRN:\>vm_list
>
> VM_UUID                          VM_ID VM_NAME                          VM_STATE
> ================================ ===== ================================ ======== [...]

 Do you have any message from the Debian VM when you were starting it alone using the original script? Because there is no GPU, we need to make sure that the Debian VM sees a serial port **and** writes to it (inc. starting a login service on it).

> Now I move to a WHL platform(ACRN V2.3, Industry Scenario), just have one disk, maybe cannot use launch_hard_rt_vm.sh.

I think you could still do that, by replacing the `-s 2,passthru,02/0/0 \` line with `-s 2,virtio-blk,./debian10-40G.img \` and commenting out this section:
```
# NVME pass-through
echo ${passthru_vpid["nvme"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["nvme"]} > /sys/bus/pci/devices/${passthru_bdf["nvme"]}/driver/unbind
echo ${passthru_bdf["nvme"]} > /sys/bus/pci/drivers/pci-stub/bind
```

> 
> I tried the launch_xenomai.sh, and have another issue.
> sudo ./launch_xenomai.sh -f ./preempt-rt-32030.img -k ./linux-4.19.59-xenomai-3.1-acrn+-x86.tar.gz

Can you send us the output of `fdisk -l  ./preempt-rt-32030.img`?

SIMING29, 2021-04-06:
1. launch debian user vm

> Do you mean you changed the UUID that was indicated in my launch script above? Can you share that latest script (that may be easier for me to look at).

added the -U parameter in your first script.   [launch_debian.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/6263139/launch_debian.sh.txt)
When I ran the script alone, got the same message as before.

I added  `“-s 5,virtio-console,@stdio:stdio_port \”`  in the launch_debian.sh script 
and added  `"console=tty0 console=hvc0 console=ttyS0"` in debian kernel command line.
still cannot login on the terminal, but can login in through ACRN debug uart.
```
ACRN:\>vm_console 3

----- Entering VM 3 Shell -----
s=remount-ro
         Starting Apply Kernel Variables...
[  OK  ] Started Remount Root and Kernel File Systems.
...
...
         Starting Update UTMP about System Runlevel Changes...
[  OK  ] Started Update UTMP about System Runlevel Changes.

Debian GNU/Linux 10 acrn-debian ttyS0
acrn-debian login: root
Password: 
Last login: Tue Apr  6 05:07:15 EDT 2021 on ttyS0
Linux acrn-debian 4.19.59-rt24 #1 SMP PREEMPT RT Mon Mar 22 22:08:14 EDT 2021 x86_64
```

**How to set the launch script or debian to make the debian uos can start like CLR uos which can login on the terminal?**

2. launch rtvm
> Can you send us the output of `fdisk -l ./preempt-rt-32030.img`?

the image file is download accroading to [the address in tutorials](https://projectacrn.github.io/latest/tutorials/using_xenomai_as_uos.html#launch-the-rtvm).
```
$ fdisk -l ./preempt-rt-32030.img 
Disk ./preempt-rt-32030.img: 8.6 GiB, 9169755648 bytes, 17909679 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 29FE431C-1800-4AAA-8478-9615A1ECA67B

Device                    Start      End  Sectors  Size Type
./preempt-rt-32030.img1    2048  1046527  1044480  510M EFI System
./preempt-rt-32030.img2 1046528  1114111    67584   33M Linux swap
./preempt-rt-32030.img3 1114112 17891327 16777216    8G Linux root (x86-64)

```
[lspci.txt](https://github.com/projectacrn/acrn-hypervisor/files/6263530/lspci.txt)

SIMING29, 2021-04-07:
The error message of using launch_hard_rt_vm.sh to start uos is the same as using launch_xenomai.sh
[launch_hard_rt_vm.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/6268443/launch_hard_rt_vm.sh.txt)
```
[  131.626764] random: crng init done
[  131.626769] random: 7 urandom warning(s) missed due to ratelimiting
[  187.482028] smpboot: CPU 1 is now offline
[  187.482492] vhm: try to offline cpu 1 with lapicid 2
[  187.496579] smpboot: CPU 2 is now offline
[  187.497175] vhm: try to offline cpu 2 with lapicid 4
[  187.520163] smpboot: CPU 3 is now offline
[  187.520643] vhm: try to offline cpu 3 with lapicid 6
[  187.544660] vhm_dev_open: opening device node
[  187.551666] vhm-ioreq: init request buffer @ 000000003e97d773!
[  187.551672] vhm-ioreq: created ioreq client 1 for ioeventfd-2
[  187.554572] ACRN vhm ioeventfd init done!
[  187.554598] ACRN vhm irqfd init done!
[  187.554600] vhm: VM 2 created
[  187.851031] vhm-ioreq: created ioreq client 2 for acrndm
[  189.522953] vhm: failed to set irqline!
[  189.534924] vhm: failed to set irqline!
[  189.601601] vhm: failed to set irqline!
[  191.090284] vhm: failed to inject!
[  191.102169] vhm: failed to inject!
[  191.113926] vhm: failed to inject!
[  191.133233] vhm: failed to inject!

```

SIMING29, 2021-04-08:
There will be warning messages in  ACRN debug uart when running launch_hard_rt_vm.sh 
```ACRN:\>[777810024us][cpu=0][vm0:vcpu0][sev=3][seq=236]:ret=-1 hypercall=0x80000080 failed in vmcall_vmexit_handler
[777821730us][cpu=0][vm0:vcpu0][sev=3][seq=237]:ret=-1 hypercall=0x80000080 failed in vmcall_vmexit_handler
[777833459us][cpu=0][vm0:vcpu0][sev=3][seq=238]:ret=-1 hypercall=0x80000080 failed in vmcall_vmexit_handler
[777845108us][cpu=0][vm0:vcpu0][sev=3][seq=239]:ret=-1 hypercall=0x80000080 failed in vmcall_vmexit_handler
[779553697us][cpu=0][vm0:vcpu0][sev=3][seq=243]:ret=-1 hypercall=0x80000025 failed in vmcall_vmexit_handler
[779607286us][cpu=2][vm2:vcpu0][sev=3][seq=244]:vlapic: Start Secondary VCPU1 for VM[2]...
[779621896us][cpu=0][vm0:vcpu0][sev=3][seq=246]:ret=-1 hypercall=0x80000025 failed in vmcall_vmexit_handler

[780484412us][cpu=2][vm2:vcpu0][sev=3][seq=273]:vioapic: WARNING: invalid vpic wire mode change
[780494579us][cpu=2][vm2:vcpu0][sev=3][seq=274]:vioapic: WARNING: invalid vpic wire mode change
[780504834us][cpu=2][vm2:vcpu0][sev=3][seq=275]:vioapic: WARNING: invalid vpic wire mode change
[780643409us][cpu=2][vm2:vcpu0][sev=3][seq=276]:vioapic: WARNING: invalid vpic wire mode change
[780653576us][cpu=2][vm2:vcpu0][sev=3][seq=277]:vioapic: WARNING: invalid vpic wire mode change
[780797875us][cpu=2][vm2:vcpu0][sev=3][seq=278]:vioapic: WARNING: invalid vpic wire mode change
[780808041us][cpu=2][vm2:vcpu0][sev=3][seq=279]:vioapic: WARNING: invalid vpic wire mode change
[780949143us][cpu=2][vm2:vcpu0][sev=3][seq=280]:vioapic: WARNING: invalid vpic wire mode change
[780959311us][cpu=2][vm2:vcpu0][sev=3][seq=281]:vioapic: WARNING: invalid vpic wire mode change

[781118816us][cpu=0][vm0:vcpu0][sev=3][seq=282]:ret=-1 hypercall=0x80000023 failed in vmcall_vmexit_handler
[781131523us][cpu=0][vm0:vcpu0][sev=3][seq=283]:ret=-1 hypercall=0x80000023 failed in vmcall_vmexit_handler
[781144428us][cpu=0][vm0:vcpu0][sev=3][seq=284]:ret=-1 hypercall=0x80000023 failed in vmcall_vmexit_handler
[781164192us][cpu=0][vm0:vcpu0][sev=3][seq=285]:ret=-1 hypercall=0x80000023 failed in vmcall_vmexit_handler
```

SIMING29, 2021-04-09:
> How to set the launch script or debian to make the debian uos can start like CLR uos which can login on the terminal?

I reconfigured the scenario and launch xml using ACRN configuration editor.Then it worked. I think the possible key factor is the setting of uos_type, which I set before is WINDOWS, now it's GENERIC LINUX.

zhangshuai75, 2021-04-23:
> The error message of using launch_hard_rt_vm.sh to start uos is the same as using launch_xenomai.sh [...]

Use this patch to enable polling mode in linux-stable-xenomai, or git clone the newest linux-stable-xenomai.
[Merge pull request #2 from tw4452852/virtio-polling-mode · intel/linux-stable-xenomai@224f688 · GitHub](https://github.com/intel/linux-stable-xenomai/commit/224f68869f267cd9df60399d32553431e71afc8a)

closed 2021-04-23 by zhangshuai75