projectacrn/acrn-hypervisor #5895: hv: Hide HLAT feature from guests

yfliuuu, 2021-03-31:
For platform with HLAT (Hypervisor-managed Linear Address Translation) capability, the hypervisor shall hide this feature from its guests.

yfliuuu, 2021-03-31:
[External_System_ID] ACRN-6627

NanlinXie, 2021-04-27:
Doc Impact Analysis: 

- [ ] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

NanlinXie, 2021-04-27:
Minor change from developer view.

closed 2021-07-12 by NanlinXie