projectacrn/acrn-hypervisor #5897: [EHL] Guest OS system slowdown when number IOAPIC RTEs !=24 and fail to boot if =37

jenmwms, 2021-03-31:
**Describe the bug**
Using Zephyr RTOS as Guest OS in pre-launch mode of hybrid-rt scenario, discovered asymmetric behavior for different values of configuring the number of IOAPIC RTEs. If hardcoded to use 24 RTEs, tests run as anticipated. Recently, dynamic RTE support for IOAPIC driver was implemented and we appropriately retrieve the value from ACRN vIOAPIC, which is 48 supported RTEs. 

However, we found the system takes significantly longer to complete executing related to interrutps. After testing init of various hardcoded RTEs, we find especially strange behavior at number of RTEs = 37. Init'ng 37 causes ACRN to not boot Zephyr image anymore. 

So, to sum it up, if we initialize IO-APIC RTE 0-23, everything works as anticipated, but if we write the same value into index 24 (which we do not use), the system bogs down.

Are there reserved addresses which we should not init in the 48 supported RTEs, or perhaps double-use of indices? Appears there a bug in the vIOAPIC emulation, please investigate.

**Platform**
EHL

**Codebase**
OS: Linux,
Toolchain: Zephyr SDK-0.12.3
Commit SHA or Version: zephyr github mainline: 1b99f0e 
ACRN version: acrn-2020w42.1-180000p (5289a3eb9)

**Scenario**
Hybrid
Pre-launch Zephyr RTOS as Guest OS

**To Reproduce**
1. Make sure BIOS configuration is correct for ACRN, and include/arch/x86/memory.ld line 44, LOCORE_BASE modified from 0x1000 to 0x8000.

2. Run this command from the zephyrproject workspace, where .exp is expect script to step through the ACRN VM Console to boot Zephyr image:
twister -p acrn_ehl_crb --device-testing --device-serial-pty="/opt/remotehw/acrn-test-pty.exp,ehlsku11" --west-flash="/opt/remotehw/remotehw-x86-acrn.sh,ehlsku11" -vv -T tests/kernel/
(you can pick one to run, for example: -T tests/kernel/mutex/mutex_api to reproduce it. )

3. Check handler.log in twister.out in directory to see the error log.
You can see a line "Timed out while monitoring serial output on"

**Actual behavior**
```
twister - DEBUG - DEVICE: ACRN:\>vm_console 0
twister - DEBUG - DEVICE:
twister - DEBUG - DEVICE: ----- Entering VM 0 Shell -----
twister - DEBUG - DEVICE: *** Booting Zephyr OS build zephyr-v2.5.99-ww11.3  ***
twister - DEBUG - DEVICE: Running test suite mutex_api
twister - DEBUG - DEVICE: ===================================================================
twister - DEBUG - DEVICE: START - test_mutex_lock_unlock
twister - DEBUG - DEVICE: PASS - test_mutex_lock_unlock in 0.0 seconds
twister - DEBUG - DEVICE: ===================================================================
twister - DEBUG - DEVICE: START - test_mutex_reent_lock_forever
twister - DEBUG - DEVICE: access resource from main thread
twister - DEBUG - Timed out while monitoring serial output on acrn_ehl_crb
twister - DEBUG - Process /opt/remotehw/acrn-test-pty.exp,ehlsku11 terminated outs: None errs None
twister - DEBUG - run status: acrn_ehl_crb/tests/kernel/mutex/mutex_api/kernel.mutex timeout
twister - ERROR - acrn_ehl_crb              tests/kernel/mutex/mutex_api/kernel.mutex          ^[[31mFAILED^[[39m: Timeout
twister - ERROR - see: ^[[33m/home/chenp1/work/zephyr-intel-ehl/zephyr/twister-out/acrn_ehl_crb/tests/kernel/mutex/mutex_api/kernel.mutex/handler.log^[[39m
```

**Expected behavior**
No obvious system slowdown or asymmetric behavior of all 48 of the supported RTEs at initialization. 
Observe expected behavior by setting ioapic_rtes=24 instead of the dynamic lookup from MRE which gives us 48.
Tests complete without issue in the 60s timeout for serial output on the test case.

**Additional context**
More info https://github.com/zephyrproject-rtos/zephyr/issues/33593. After git bisect, we found it failed start from the commit: 915f4ac which implemented getting the ioapic_rtes dynamically.

jsun26intel, 2021-04-01:
have you tested on native ehl?

jsun26intel, 2021-04-02:
I've replied in zephyr github issue,  and let Zephyr QA team have a try with CONFIG_APIC_TIMER_IRQ=48.
I guess there is a IRQ overlap between IOAPIC and APIC TIMER.

andyross, 2021-04-02:
FWIW: the APIC timer interrupt doesn't use an IO-APIC line architecturally, it's always CPU-local.

The suspicion here is, indeed, that ACRN is exposing interrupt inputs to the IO-APIC that aren't external devices and/or shouldn't be initialized by the OS.  All Zephyr is doing with those lines is setting them to a known "idle" state (per the code: edge triggered, active high, masked, fixed delivery to a logical destination with zero CPU bits set).  That configuration should cause the device to be silent, instead we're apparently seeing some kind of interrupt storm.

jenmwms, 2021-04-07:
> have you tested on native ehl?

Yes, this is tested on native ehl. This issue is not observed.

jenmwms, 2021-04-07:
> I've replied in zephyr github issue, and let Zephyr QA team have a try with CONFIG_APIC_TIMER_IRQ=48.
> I guess there is a IRQ overlap between IOAPIC and APIC TIMER.

Thank you! This does help give a definition that works reliably with Zephyr driver's support of dynamic RTEs, which retreives 48 as expected. However, this is more of a patch than a fix. There is still something going on as Andy described in comment above. Please consider this still a potential issue that we may be evading rather than resolving.

andyross, 2021-04-08:
Ah, OK.  So this is looking like a Zephyr bug then.

Our APIC code configures a "IRQ" in a funny space -- APIC interrupts are delivered to CPU vectors, not IO-APIC lines.  So the vectors get encoded such that the vector indexes lie immediately after the IO-APIC RTEs.  But when the number of RTEs changes (as it does here under ACRN) we need to move the APIC "IRQ" to match so it encodes the same vector.  In this case it ended up too low and inside the IO-APIC space, so the interrupt controller driver assumed it had to connect the timer driver to an IO-APIC interrupt that had nothing else connected to it, and so nothing happened, I'm assuming?

Regardless, this is a design bug on our side: we should specify APIC interrupt targetting in a way that isn't dependent on unrelated areas of the system like "how many inputs are there to the IO-APIC?".

jsun26intel, 2021-04-09:
@andyross  @jenmwms 
please double check zephyr's implementation.
EHL native doesn't use APIC timer, it use HPET timer by default, whereas ACRN has no vHPET implementation, so we have to use APIC timer instead. That's why the issue could not be reproduced on EHL native. Maybe you can have a try on APIC timer with SMP disabled.

On Zephry's apic timer implementation (drivers/timer/apic_timer.c), it would enable the hardcoded APIC IRQ(24), thus the ioapic pin would be enabled by below function:

void arch_irq_enable(unsigned int irq)
{
	if (IS_IOAPIC_IRQ(irq)) {
		z_ioapic_irq_enable(irq);
	} else {
		z_loapic_irq_enable(irq - z_loapic_irq_base());
	}
}

jsun26intel, 2021-04-09:
oh, sorry. I missed the comments from @andyross .

fuzhongl, 2021-04-13:
Closed as it isn't ACRN bug.

closed 2021-04-13 by fuzhongl