projectacrn/acrn-hypervisor #5898: Can't Create a Windows 7 Image

SIMING29, 2021-04-01:
WHL platform, ACRN2.3, Industry scenario
I have created two versions of Windows10 image, both of them worked well.
When I created a windows7 image, the VM stuck at "starting Windows".

[install_win7.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/6240498/install_win7.sh.txt)
[image]

```
acrn@acrn-WL37:~/vm-win$ sudo ./install_win7.sh 
[sudo] password for acrn: 
cpu1 online=1
cpu2 online=1
cpu3 online=1
SW_LOAD: get ovmf path /home/acrn/vm-win/OVMF.fd, size 0x200000
vm_create: win_vm1
VHM api version 1.0
vm_setup_memory: size=0x100000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/1 page size:0x200000
level 1 free/need pages:12/4 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 1 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x80000000
total_size 0x200000000

mmap ptr 0x0x7f06712ed000 -> baseaddr 0x0x7f0680000000
mmap 0x80000000@0x7f0680000000
touch 2 pages with pagesz 0x40000000
mmap 0x80000000@0x7f07c0000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7f077fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x80000000
vm_init_vdevs
No correct pm notify channel given
polling 34...
Listening 34...
pci init hostbridge
pci init lpc
pci init passthru
pci init virtio-blk
pci init ahci
pci init ahci
pci init passthru
pci init virtio-net
pci init igd-lpc
tpm: init_vtpm2:Invalid socket path!
                                    /tmp/dm.Xs4Brwf     24:     Device (PCI0)
                                                                             Warning  3073 -        Multiple types ^  (Device object requires either a _HID or _ADR, but not both)

       /tmp/dm.Xs4Brwf    813:     Processor (CPU0, 0x00, 0x00000000, 0x00) {}
                                                                              Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                /tmp/dm.Xs4Brwf    814:     Processor (CPU1, 0x01, 0x00000000, 0x00) {}
                                                                                       Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                         acrn_sw_load
                                     SW_LOAD: partition blob /home/acrn/vm-win/OVMF.fd size 0x200000 copied to addr 0x7f077fe00000
                                                                                                                                  SW_LOAD: build e820 9 entries to addr: 0x7f06800ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x0000000000100000, size 0x000000003b700000,  type 0x1
SW_LOAD: entry[2]: addr 0x000000003b800000, size 0x0000000004004000,  type 0x2
SW_LOAD: entry[3]: addr 0x0000000040080000, size 0x0000000000800000,  type 0x2
SW_LOAD: entry[4]: addr 0x0000000040880000, size 0x000000003f780000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000140000000, size 0x0000000080000000,  type 0x1
SW_LOAD: entry[7]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: entry[8]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu

```

gvancuts, 2021-04-02:
Hi @SIMING29 , we have never tried with Windows 7 AFAIK... @NanlinXie can you confirm this?

fuzhongl, 2021-04-06:
@gvancuts  You are right! We have never tried with Windows 7.
Thanks!

zhangshuai75, 2021-04-23:
not success with Win7

closed 2021-04-23 by zhangshuai75