projectacrn/acrn-hypervisor #5903: [Doc] Mismatch in text and picture in ACRN introduction documentation

NanlinXie, 2021-04-02:
There's mismatch in text and Figure 8 for partition mode description, Figure 8 tells safety/rtvm for both guest, but description tells one is safety and another is user VM.

See ACRN webpage https://projectacrn.github.io/latest/introduction/index.html#logical-partition-scenario, which cause user confused about that.

gvancuts, 2021-04-02:
Good catch, thanks @NanlinXie !

Looking through our various `logical_partition.xml` scenario files, they all only use 2 pre-launched standard VMs (`PRE_STD_VM`). The introductory text should talk rather generically about these different scenario in my opinion. So how about I change the text to indicate that in such scenario one could launch different kinds of VMs (Safety, RTVM, standard), the whole point of this scenario is that both are pre-launched, there is no Service VM and hence no `acrn-dm` on that system.

I'm also tempted to add a note stating that the examples currently only define 2 Pre-launched Standard VM.

Thoughts? @dbkinder @fitchbe would love to hear your opinions too

fitchbe, 2021-04-02:
The proposed change to the text to avoid being unnecessarily specific sounds really good to me. Not as sure about the added note, because notes with the word "currently" in them can become out of date. :-) Let's see what @dbkinder thinks.

dbkinder, 2021-04-02:
Agree with Ben here.  We avoid using "currently" or "in the next release" or such wording that becomes confusing.  @gvancuts can you submit a PR to address this...

gvancuts, 2021-04-06:
> @gvancuts can you submit a PR to address this...

Done, please take a look at #5906

dbkinder, 2022-01-26:
implemented

closed 2022-01-26 by dbkinder