projectacrn/acrn-hypervisor #5907: hv: wrong condition to warn DRHD's INCLUDE_PCI_ALL flag

stanleyintel, 2021-04-07:
**Describe the bug**
In hypervisor/acpi_parser/dmar_parse.c, the parse_dmar_table() shall not print out a warning message when parsing the last DRHD. Instead, it should print out a warning message when previous DRHD already set the DRHD_FLAG_INCLUDE_PCI_ALL_MASK flag. Specifically, the following if-condition.
(!is_include_all_emulated) is incorrect.

        if (!is_include_all_emulated) {
            pr_warn("DRHD with INCLUDE_PCI_ALL flag is NOT the last one!");
        }

**Platform**
All

**Codebase**
master

**Scenario**
All

fuzhongl, 2021-05-03:
@NanlinXie Please help to assign developer to check this issue.
Thanks!

lifeix, 2022-03-22:
yes,  we have this problem.

fuzhongl, 2022-03-22:
[External_System_ID] ACRN-8693

fuzhongl, 2022-04-01:
@stanleyintel Patch is merged into master branch.
Please help to check.
Thanks!

closed 2022-04-14 by fuzhongl