projectacrn/acrn-hypervisor #5913: DM:  RAM and MMIO address change for post-launched VM

taoyuhong, 2021-04-09:
change high RAM start address to 4GB for post-launched VM
change 64-bit mmio address space to 256G-512G for post-launched VM

taoyuhong, 2021-04-09:
[External_System_ID] ACRN-6526

NanlinXie, 2021-04-27:
Doc Impact Analysis: 

- [x] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

taoyuhong, 2021-05-08:
These changes  have no impact with HLD,

NanlinXie, 2021-05-26:
@gvancuts , could we change to doc_done?

gvancuts, 2021-05-26:
> @gvancuts , could we change to doc_done?

It looks like these patches at least affect the Device Model HLD, e.g. the part where the e820 memmap is described.

I haven't done a very thorough check but it's the only one I found that needs updating.

NanlinXie, 2021-06-07:
@JasonChenCJ, any comment?

JasonChenCJ, 2021-06-07:
These two patches are actually not directly related with MMIO device pass-thru..

gvancuts, 2021-06-18:
These do not match for example:
* https://github.com/projectacrn/acrn-hypervisor/blob/daccad241faf0bf20341beb49c4ae9390956845b/devicemodel/core/sw_load_common.c#L42-L64
* https://projectacrn.github.io/latest/developer-guides/hld/hld-devicemodel.html

NanlinXie, 2021-06-18:
@JasonChenCJ  and @taoyuhong , any update according to @gvancuts's comment?

taoyuhong, 2021-06-19:
"change high RAM start address to 4GB for post-launched VM" is related change, but that change matches HLD:
 * 5:   0xe00_00000 -     fff_fffff   (reserved)   512MB
 * 6:   1_000_00000 -     highmem     RAM          highmem-4G
 */ 
before "change high RAM start address to 4GB for post-launched VM", highmem start at 5G, that does not match HLD, need update. But after #5913, we change highmem back to 4G, match HLD again

taoyuhong, 2021-06-21:
The e820 layout has changed, does not match HLD, need update HLD. I just mean these mismatch does not caused by mmio device passthrough  patches. 
Mismatch has happened since commit 643513f3d484454fc5d0ca627e3b94da952c7198
Author: Peter Fang <peter.fang@intel.com>
Date:   Fri Mar 15 01:59:56 2019 -0700
 
And after that it keep changed several times. I am not sure if it will change again soon.  So I don't know if we should update HLD now.

gvancuts, 2021-06-21:
> 
> 
> The e820 layout has changed, does not match HLD, need update HLD. I just mean these mismatch does not caused by mmio device passthrough patches.
> Mismatch has happened since commit [643513f](https://github.com/projectacrn/acrn-hypervisor/commit/643513f3d484454fc5d0ca627e3b94da952c7198)
> Author: Peter Fang [peter.fang@intel.com](mailto:peter.fang@intel.com)
> Date: Fri Mar 15 01:59:56 2019 -0700

Oh, I see. Thanks for the clarification.

> 
> And after that it keep changed several times. I am not sure if it will change again soon. So I don't know if we should update HLD now.

We should. And we should again the next time it changes ;-)

liuhang-bit, 2021-08-25:
https://github.com/projectacrn/acrn-hypervisor/pull/6404 this PR fix the issue

NanlinXie, 2021-08-25:
@gvancuts , please confirm whether we could change to doc:done.

gvancuts, 2021-08-25:
> 
> 
> @gvancuts , please confirm whether we could change to doc:done.

Yes

amyreye, 2022-02-17:
@NanlinXie @dbkinder Can we close this issue?