projectacrn/acrn-hypervisor #5915: Hypervisor PTM Support

NanlinXie, 2021-04-09:
The Hypervisor shall provide the hypervisor specific functionality needed for PTM(Precision Time Management) to function on PTM enabled PCIE devices that are passed through or mediated.

NanlinXie, 2021-04-09:
[External_System_ID] ACRN-5896

rliu88, 2021-04-22:
For post launch:
To use, to "enable-ptm" when pass through device, example in launch script:
/home/labuser/RL/bin/acrn-dm -A -m $mem_size -s 0:0,hostbridge \
  -s 3,virtio-blk,/home/labuser/RL/ubuntu_images/uos-test.img \
  -s 4,virtio-net,tap0 \
  -s 5,virtio-console,@stdio:stdio_port \
  -s 6,passthru,a9/00/0,**enable_ptm** \
  --ovmf /usr/share/acrn/bios/OVMF.fd \
}

Sample output from guest os when ptm is enabled (check using dmesg):
uos@uos-Standard-PC-i440FX-PIIX-1996:~$ dmesg | grep -i ptm
[    1.555284] <PTM-init>pci_ptm_init: 00:00.00, ispcie=1, type=0x4
[    1.555306] <PTM-init>no upstream bridge
[    1.555356] <PTM-init> Cannot find PTM ext cap.
[    1.561311] <PTM-init>pci_ptm_init: 00:03.00, ispcie=0, type=0x0
[    1.567146] <PTM-init>pci_ptm_init: 00:04.00, ispcie=0, type=0x0
[    1.572983] <PTM-init>pci_ptm_init: 00:05.00, ispcie=0, type=0x0
[    1.718038] <PTM-init>pci_ptm_init: 00:06.00, ispcie=1, type=0x4
[    1.718056] <PTM-init>no upstream bridge
[    1.722034] <PTM-init> ptm is ptm_root.
[    1.723033] <PTM-init> Condition-2: ptm is enabled.
[    1.723052] pci 0000:00:06.0: PTM enabled (root), 4ns granularity
[    1.766438] <PTM-init>pci_ptm_init: a9:00.00, ispcie=1, type=0x0
[    1.766456] <PTM-init>Ethernet controller upstream bridge port: 00:06.00
[    5.715000] igc_probe enable ptm.
[    5.715068] <PTM-en>pci_enable_ptm: a9:00.00, ispcie=1, type=0x0
[    5.715294] <PTM-en> ptm is enabled on endpoint device.
[    5.715371] igc 0000:a9:00.0: PTM enabled, 4ns granularity

NanlinXie, 2021-04-27:
Doc Impact Analysis: 

- [x] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

gvancuts, 2021-05-07:
I suspect this HLD will need to be updated too: https://projectacrn.github.io/latest/developer-guides/hld/hv-dev-passthrough.html

rliu88, 2021-05-10:
I reviewed the above document and since we didn't change how passthru device work, we don't need to update above document.  We do need a document similar to emulated host bridge and that is covered by HLD.  Let me know if I miss anything.

rliu88, 2021-05-10:
Where should I put "user guide" for ptm feature?  There are a few steps that a user need to do in order to use PTM

rliu88, 2021-05-11:
Here is HLD design for virtual root port.  Please review.  Thanks.

PCIE Root Port Emulation
Overview
In order to support PTM feature without changing guest’s kernel implementation, ACRN adds a pci-e root port emulation in hypervisor.  To simplify the implementation,  virtual root port only supports the most basic pci-e configuration and operation, in addition to PTM capabilities.  The virtual root port will be used in both post-launch and pre-launch (if applicable) scenarios.  
Guest PCI-e Hierarchy
When user enables ptm while passing thru a device to post launch guest,  the pass thru device is connected to virtual root port instead of host bridge.  Here is an example of launch script that enables ptm on pass thru ethernet card:
/home/labuser/RL/bin/acrn-dm -A -m $mem_size -s 0:0,hostbridge \
  -s 3,virtio-blk,/home/labuser/RL/ubuntu_images/uos-test.img \
  -s 4,virtio-net,tap0 \
  -s 5,virtio-console,@stdio:stdio_port \
  -s 6,passthru,a9/00/0,enable_ptm \
  --ovmf /usr/share/acrn/bios/OVMF.fd \
}
And here is the bus hierarchy from guest
-[0000:00]-+-00.0  Network Appliance Corporation Device 1275
           +-03.0  Red Hat, Inc. Virtio block device
           +-04.0  Red Hat, Inc. Virtio network device
           +-05.0  Red Hat, Inc. Virtio console
           \-06.0-[01]----00.0  Intel Corporation Device 15f2

rliu88, 2021-05-11:
PTM HLD.  Please review.  Thanks.

Enable PTM in ACRN
PCI-e spec provides Precision Time Measurement (PTM) mechanism to enable time coordination and synchronization of events across multiple pci components with independent local time clocks within the same system.  
Intel supports PTM on several of its systems and devices, such as PTM root capabilities support on Whiskey Lake and Tiger Lake PCI-e root ports, and PTM device support on I225-V ethernet controller.
For further details on PTM, please refer to PCI-e specification.

PTM Support in ACRN
ACRN emulates a simple PTM hierarchy.  ACRN enables PTM in guest virtual machine if user sets “enable_ptm” option when passing through a device to a post launch vm.  
 

PTM High-level design in ACRN
In post launch scenarios, user enables PTM by setting “enable_ptm” option when pass through a device to vm.  Device model then checks whether the pass through device supports PTM requestor capabilities and whether the corresponding root port supports PTM root capabilities and performs some other sanity checks.
If an error is detected during these checks, the error will be reported to user and ACRN is not going to enable PTM on guest, however, this doesn’t prevent user from launching guest and passing through the device to guest.  If no error is detected, device model will use add_vdev hypercall to add a virtual root port (acts as PTM root) to guest before passing through the device to guest, and the passthru device will connect to the virtual root port instead of the virtual host bridge.  
 
How to enable PTM in post-launch scenario
In order to use PTM in virtualized environment, it is important to make sure that PTM is supported and enabled on baremetal machine first.
Suppose PTM is supported and enabled on host machine, follow these steps to enable it on post launch guest:
1.	Make sure that PTM is enabled in guest kernel.  In Linux kernel, for example, set “CONFIG_PTM=on”.
2.	Not every pci device supports PTM.  One example is I225-v ethernet card.  If pass thru i225-v ethernet card to guest,  make sure guest uses a version of igc driver that supports PTM.
3.	In launch script, add “enable_ptm” option on the pass-thru device.  For example:
/home/labuser/RL/bin/acrn-dm -A -m $mem_size -s 0:0,hostbridge \
  -s 3,virtio-blk,/home/labuser/RL/ubuntu_images/uos-test.img \
  -s 4,virtio-net,tap0 \
  -s 5,virtio-console,@stdio:stdio_port \
  -s 6,passthru,a9/00/0,enable_ptm \
  --ovmf /usr/share/acrn/bios/OVMF.fd \
}
4.	Check that PTM is correctly enabled on guest
Upon successfully enabled PTM, example pci bus hierarchy on guest:
>sudo lspci -tv
          -[0000:00]-+-00.0  Network Appliance Corporation Device 1275
           +-03.0  Red Hat, Inc. Virtio block device
           +-04.0  Red Hat, Inc. Virtio network device
           +-05.0  Red Hat, Inc. Virtio console
           \-06.0-[01]----00.0  Intel Corporation Device 15f2
>sudo lspci -vv (Only relevant output are shown)
00:00.0 Host bridge: Network Appliance Corporation Device 1275
00:06.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev 02) (prog-if 00 [Normal decode])
….
        Capabilities: [100 v1] Precision Time Measurement
                PTMCap: Requester:- Responder:+ Root:+
                PTMClockGranularity: 4ns
                PTMControl: Enabled:+ RootSelected:+
                PTMEffectiveGranularity: 4ns
        Kernel driver in use: pcieport
01:00.0 Ethernet controller: Intel Corporation Device 15f2 (rev 01)
…
        Capabilities: [1f0 v1] Precision Time Measurement
                PTMCap: Requester:+ Responder:- Root:-
                PTMClockGranularity: 4ns
                PTMControl: Enabled:+ RootSelected:-
                PTMEffectiveGranularity: 4ns
        Kernel driver in use: igc

rliu88, 2021-05-11:
Is there a way to include pictures?

dbkinder, 2021-06-15:
#6110 provides the documentation (and is already merged)

dbkinder, 2021-06-16:
@NanlinXie I was going to close this issue (since the doc is done) but I see there were more code changes being made so I've left it open.

dbkinder, 2021-07-12:
PTM tutorial was removed for v2.5 (PR #6258 ) so a new PR is needed to put it back.

rliu88, 2021-07-12:
I didn’t know that PTM is not in 2.5 release.  There is nothing to fix.  The failed test is PTP and even that is caused by ART not enabled in VM.  I updated the test with root cause and it is verified by QA.  All codes are checked into master.

NanlinXie, 2021-07-15:
@dbkinder , please help to close this issue once the document is published.

rliu88, 2021-07-16:
@dbkinder I think we should add a section talking about use cases of PTM in PTP.  The most important thing is that we can show the correct configuration to enable PTP.  What do you think?  I can give you a draft next week.

dbkinder, 2021-07-16:
@rliu88  PTM tutorial is merged back on master now (PR #6302) New information about use cases would be great - either email me with new material or submit a PR with your additions and I'll review.

rliu88, 2021-07-22:
@Kinder, David ***@***.***> I just emailed you my draft.  Please let me know if you have suggestion.  Thanks!

NanlinXie, 2021-11-08:
We are aligned not to publish this document on github.

dbkinder, 2022-01-26:
@NanlinXie Since we're not publishing documentation for this feature, can the issue be closed?

amyreye, 2022-02-16:
@NanlinXie @dbkinder Can we close this issue?

closed 2022-03-04 by fuzhongl