projectacrn/acrn-hypervisor #5917: Virtualization RDT with CAT/CDP - Single VM

NanlinXie, 2021-04-09:
The Hypervisor and ServiceOS shall be able to enable virtual CAT (Cache Allocation Tech) and virtual CDP (Cache Data Prioritization) for a single Guest VM.

NanlinXie, 2021-04-09:
[External_System_ID] ACRN-6101

NanlinXie, 2021-04-09:
Doc Impact Analysis: 

- [ ] Does the change introduce any hypervisor configuration data?	

- [ ] Does the change add or modify any kernel module parameter?	

- [ ] Does the change add or modify any DM command line option?	

- [ ] Does the change add or modify any command line option in acrn-tools?	

- [ ] Does the change add or modify any command to acrn shell (i.e. the HV UART shell for debugging)?	

- [ ] Does the change modfy the acrn-hypervisor/Makefile	

- [ ] Does the change require additional library to be installed in yocto?	

- [ ] Does the change add or modify any service configuration file (under misc/packaging) in acrn-hypervisor repo?	****

gvancuts, 2021-05-07:
@NanlinXie , as I was looking into this one, I realized that the current `doc/tutorials/rdt_configuration.rst`:
* https://projectacrn.github.io/latest/tutorials/rdt_configuration.html#configure-rdt-for-vm-using-vm-configuration (esp. the extract of config files)
* Add `sudo apt install hwloc-nox` as a note so the user has the `lstopo-no-graphics` installed
* Remove the need to be root for `lstopo-no-graphics` (`#` shell prompt)
* Update step 6 that give simplified build instructions (`make` command is no longer valid and it talks about `acrn.efi`)

NanlinXie, 2021-05-24:
Postpone to future release.

amyreye, 2022-02-17:
@NanlinXie @dbkinder Can we close this issue?

closed 2022-03-04 by fuzhongl