projectacrn/acrn-hypervisor #5918: Hybrid cores support in hypervisor

NanlinXie, 2021-04-09:
The hypervisor shall be able to do enumeration and boot all big and small cores in its context.

NanlinXie, 2021-04-09:
[External_System_ID] ACRN-6632

NanlinXie, 2021-04-27:
Doc Impact Analysis: 

- [x] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

NanlinXie, 2021-04-27:
There's one limitation for configuration: only support assign same cores to a VM.

gvancuts, 2021-05-07:
> There's one limitation for configuration: only support assign same cores to a VM.

Where should this be mentionned?

Should this HLD be updated as a result of this new feature? https://projectacrn.github.io/latest/developer-guides/hld/hv-cpu-virt.html

NanlinXie, 2021-05-26:
@lifeix , Any comments?

lifeix, 2021-05-26:
> > There's one limitation for configuration: only support assign same cores to a VM.
> 
> Where should this be mentionned?
> 
> Should this HLD be updated as a result of this new feature? https://projectacrn.github.io/latest/developer-guides/hld/hv-cpu-virt.html

Actually, we could not claim we have supported hybrid core yet. We still need to assign the same cores to a VM. In theory, this only includes the configuration tool's change. So I think there's no need to update the HLD now. Thanks.

NanlinXie, 2021-06-07:
The error descirption already updated in config tool xsd file, please check whether document is proper generated. @dbkinder

dbkinder, 2021-06-14:
@NanlinXie @junjiemao1 We have no processing that turns ``<xs:assert . . . >`` annotations into documentation.  Let's talk about how and where such assert documentation should be processed and generated.

dbkinder, 2021-06-16:
@NanlinXie I don't think we can close this issue since the the implementation (and documentation) for supporting hybrid cores is not done.

NanlinXie, 2021-07-12:
@dbkinder , I would like to put the topic "how and where such assert documentation should be processed and generated" into DX future debug discussion. This is a general topic not just this issue. For now, we don't provide the document that described such assert information.  For this issue, there's only change to the assert , thus there's no document change needed, could we close this issue?

dbkinder, 2022-01-26:
@NanlinXie Is there an update for this?

closed 2022-03-04 by fuzhongl