projectacrn/acrn-hypervisor #5920: Modularization for paging/memory part.

NanlinXie, 2021-04-09:
ACRN shall be built with multiple modules
Same functionality module shall be able to realized through different implementation
Module shall be able to configurable (not choose or use different implementation)

NanlinXie, 2021-04-09:
[External_System_ID] ACRN-6846

NanlinXie, 2021-04-27:
Doc Impact Analysis: 

- [ ] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

gvancuts, 2021-05-07:
I don't have enough information yet about these changes to determine what exactly will need to be updated from a documentation point of view. So updating the HLD at a minimum (as currently indicated) seems right. Given we're talking about build options, we will need to assess if our build instructions (tools, steps, dependencies, etc.) need to be adjusted.

closed 2021-06-16 by dbkinder