projectacrn/acrn-hypervisor #5921: [DX_task] Decouple libvirt with acrnctl and support all acrnctl’s functionality on virsh

NanlinXie, 2021-04-09:
Decouple libvirt with acrnctl and support all acrnctl’s functionality on virsh

NanlinXie, 2021-04-09:
[External_System_ID] ACRN-6920

NanlinXie, 2021-04-12:
Doc Impact Analysis: 

- [ ] Does the change require HLD update? If yes, need update HLD.

- [ ] Does the change introduce any hypervisor configuration data? If yes, need update configuration item description.

- [ ] Does the change add or modify any kernel module parameter? If yes, need update kernel module parameter description.	

- [ ] Does the change add or modify any DM command line option? If yes, need update DM cmdline option.

- [ ] Does the change add or modify any command line option in acrn-tools?	

- [ ] Does the change add or modify any command to acrn shell (i.e. the HV UART shell for debugging)?	

- [ ] Does the change modfy the acrn-hypervisor/Makefile? If yes, need check any change to meta-acrn code and doc.

- [ ] Does the change require for Service VM, library or service configuration files? If yes, need check any change to meta-acrn

gvancuts, 2021-04-12:
There are a few tutorials that reference `acrnctl`: "S5 enabling", "Kata Containers" and "OpenStack/libvirt"

Kata implementation also uses `acrnctl` so if we plan to deprecate or make the tool a debug tool like it was originally intended to be, we'll need to make adjustements to the Kata implementation and user guide.

NanlinXie, 2021-11-24:
Postpone to v2.8

dbkinder, 2022-01-26:
@NanlinXie Is this still a v2.8 target?

NanlinXie, 2022-01-28:
We will not drop acrnctl in v2.8, thus there's no change for v2.8 document.

NanlinXie, 2022-02-16:
GSG document need to be updated after this patch, we need add addtional library dependency at build time.
sudo apt-get install libcjson-dev

gvancuts, 2022-03-04:
I do not believe any documentation has been updated to reflect this change from `acrnctl` to `virsh`.

fuzhongl, 2022-03-06:
@NanlinXie The linked JIRA is in Resolved status; Could you help to check if this task is done?
Thanks!

wuxyintel, 2022-03-08:
There are a few tutorials that reference acrnctl: "S5 enabling", "Kata Containers" and "OpenStack/libvirt", Kata implementation also uses acrnctl so if we plan to deprecate or make the tool a debug tool like it was originally intended to be, we'll need to make adjustments to the Kata implementation and user guide.
@gvancuts  please help double check my comments below, thanks!
(1) In this phase, we only update libvirt to provide related virsh commands to implement acrnctl functionality. acrnctl functions still works.
(2) For document "OpenStack/libvirt", I just remove acrnctl installation since libvirt doesn't depend on it. https://github.com/projectacrn/acrn-hypervisor/pull/7166
(3) For document "S5 enabling", current libvirt can't shutdown WaaG since user VM name is hard coded in the lifecycle manager of windows, I think we can update this after this bug is fixed. The related github issue will be created soon.
(4) Kata is still use acrnctl, I think to another issue to track kata updates such as ("decouple kata with acrnctl"), BTW, kata can interact with the current DM directly now.

wuxyintel, 2022-03-24:
> > There are a few tutorials that reference acrnctl: "S5 enabling", "Kata Containers" and "OpenStack/libvirt", Kata implementation also uses acrnctl so if we plan to deprecate or make the tool a debug tool like it was originally intended to be, we'll need to make adjustments to the Kata implementation and user guide. @gvancuts please help double check my comments below, thanks! (1) In this phase, we only update libvirt to provide related virsh commands to implement acrnctl functionality. acrnctl functions still works. (2) For document "OpenStack/libvirt", I just remove acrnctl installation since libvirt doesn't depend on it. #7166 (3) For document "S5 enabling", current libvirt can't shutdown WaaG since user VM name is hard coded in the lifecycle manager of windows, I think we can update this after this bug is fixed. The related github issue will be created soon. (4) Kata is still use acrnctl, I think to another issue to track kata updates such as ("decouple kata with acrnctl"), BTW, kata can interact with the current DM directly now.
> 
> The list looks good to me. Can you tell me more about the Kata Containers status and ACRN. It used to use `acrnctl` to do the `blkrescan` operation and that was needed to run this. Have the changes for this landed yet?
> 
I remember currently kata containers fails to run on ACRN, for old Kata code, it still use "acrnctl' to do "blkrescan". I think kata code will be refined to use command monitor to do "blkrescan" after v3.0.

> Which version of `virsh` is used in our testing? Is that coming from the Ubuntu repo or it's built with the `acrn-libvirt` stuff?
The "virsh" tool should be built from "acrn-libvirt" for our testing, the version is 6.1.0.

gvancuts, 2022-05-05:
> @gvancuts , could we close this issue now ?

Not yet :-)

We need more documentation on what the tool does, how it's used, etc.

gvancuts, 2022-07-05:
Hi @fuzhongl , this can be closed once the documentation is in place, I don't recall seeing it, have I missed it?