projectacrn/acrn-hypervisor #5923: Nested Virtualization

NanlinXie, 2021-04-09:
The hypervisor shall support to run a hypervisor inside a partitioned VM, and such feature shall be configurable. The hypervisor shall support to run a hypervisor inside a partitioned VM, and such feature shall be configurable.

NanlinXie, 2021-04-09:
[External_System_ID] ACRN-6965

gvancuts, 2021-04-13:
@NanlinXie , this one will definitely require some documentation ;-)

NanlinXie, 2021-04-27:
Doc Impact Analysis: 

- [x] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

wenlingz, 2021-05-16:
[External_System_ID] ACRN-6965

NanlinXie, 2021-05-26:
doc PR: https://github.com/projectacrn/acrn-hypervisor/pull/6083
@shenfang2019 , please help to verify and review.

closed 2021-07-12 by NanlinXie