projectacrn/acrn-hypervisor #5924: [Code refine]: Refine hypercall dispatch

NanlinXie, 2021-04-09:
The permission checks in vmcall_vmexit_handler() cover multiple aspects in the same if … else if … else block, which will not scale when more kinds of checks are needed. To enhance scalability the permissions should be maintained and checked in a unified manner.

NanlinXie, 2021-04-09:
[External_System_ID] ACRN-6999

NanlinXie, 2021-04-27:
Doc Impact Analysis: 

- [ ] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

NanlinXie, 2021-04-27:
Will have impact to hypercall API error code, and will update with code PR.

gvancuts, 2021-05-05:
I don't know enough about the changes that are planned but there are some documentation about hypercalls in the HLD sections.

junjiemao1, 2021-05-12:
PR !5994, which contains both the code changes and relevant document updates, has been merged.

NanlinXie, 2021-05-12:
@junjiemao1 , could you link your doc PR here, and @szhen11 and @shenfang2019 could help on verification.

junjiemao1, 2021-05-12:
> @junjiemao1 , could you link your doc PR here, and @szhen11 and @shenfang2019 could help on verification.

It is the same PR.

dbkinder, 2021-06-15:
Indeed it is: #5994

closed 2021-06-15 by dbkinder