projectacrn/acrn-hypervisor #5928: tutorial error

iaGuoZhi, 2021-04-13:
**Describe the bug**
When I want to boot acrn through qemu, tutorial [**Enable ACRN Over QEMU/KVM**](https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html) tells me build acrn hypervisor using  tag: acrn-2020w19.5-140000p. however there is no qemu.xml  in tag: acrn-2020w19.5-140000p which is used as BOARD_FILE. 

**Codebase**

https://github.com/projectacrn/acrn-hypervisor/tree/acrn-2020w19.5-140000p/misc/acrn-config/xmls/board-xmls

gvancuts, 2021-04-13:
Good catch! (I wonder how this even passed our Q&A testing... @NanlinXie ?)

We should really take the opportunity to refresh this to be based on the latest release, i.e. `v2.4`.

iaGuoZhi, 2021-04-14:
Is there any arch-hypervisor release can build acrn over qemu? I'm eager to try ACRN over qemu.

gvancuts, 2021-04-14:
I'm in the process of trying to update this document to be based on ACRN v2.4 (the latest release as of today).

Unfortunately, ACRN v2.4 does not start. It throws the following error on the console:
```
Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Multiboot Bootloader: GRUB 2.02-2ubuntu8.21
[0us][cpu=0][(null)][sev=3][seq=1]:vlapic: Basic apicv ops is used!
detect_hardware_support, physical-address width (40) over maximum physical-address width (39)
[0us][cpu=0][(null)][sev=1][seq=2]:PANIC: init_pcpu_pre line: 132

[0us][cpu=0][(null)][sev=1][seq=3]:hardware not support!
```

@NanlinXie can you confirm what is the latest version that we have validated? Can you also ask one of the engineer to chime in as to what the problem may be with this ACRN v2.4 error on QEMU?

gvancuts, 2021-04-14:
For information, I have tried this on both Skylake and Kaby Lake processors with the same results.

iaGuoZhi, 2021-04-16:
Hi gvancuts!
I'm sorry to bother you, but I'm very eager to run ACRN so as to try out what it offers. This is a really pressing deadline for me. Therefore, As Qemu can not boot ACRN now, Would you like to help me check whether ACRN v2.4 is supported in Maxtang Intel WHL-U i7-8665U (AX8665U-A2). If it is supported, I will buy a Maxtang right now.

gvancuts, 2021-04-17:
Hi @iaGuoZhi , I don't have that board myself but I believe it works fine. @NanlinXie can you confirm?

The very latest platform that we support are the Tiger Lake-based NUCs (aka Tiger Canyon). E.g.:
* https://ark.intel.com/content/www/us/en/ark/products/212517/intel-nuc-11-pro-kit-nuc11tnhi50l.html
* https://ark.intel.com/content/www/us/en/ark/products/205603/intel-nuc-11-pro-kit-nuc11tnki5.html
There are just coming out but this is what I would recommend you buy.

NanlinXie, 2021-04-17:
Hi, @iaGuoZhi ,
1.  For the Qemu document, there's one mistake for the ACRN tag, please try ACRN v2.0 for Qemu to see whether it works for you.
2. Maxtang WHL board is working with the latest ACRN, but we suggested user to use ACRN v2.0 for WHL board for full feature.
3. As geoffroy said , we are going to recommend Tiger Lake NUC to community soon, you could join our next week's TCM to see the discussion on this topic, and then decide which board you want to buy.

iaGuoZhi, 2021-04-17:
Thanks! @gvancuts  @NanlinXie

iaGuoZhi, 2021-04-18:
Hi!
Which hw can exploit ACRN's full feature? wl10 or axwl10?
As ACRN document mentions Maxtang WL10 in Supported hardware section. https://projectacrn.github.io/2.0/reference/hardware.html#verified-platforms-according-to-acrn-usage
and AXWL10 is recommanded in getting start with ubuntu section https://projectacrn.github.io/2.0/getting-started/rt_industry_ubuntu.html
Thanks~

iaGuoZhi, 2021-05-28:
[image]

which version should we build acrnsos.img. The document doesn't say that.

gvancuts, 2021-05-28:
> Document of the latest version has not corrected this problem yet

Hopefully this is the last piece that we forgot to update. And better yet, hopefully we can do yet another update to use ACRN v2.4 or the upcoming v2.5 :-)

fuzhongl, 2021-07-06:
@iaGuoZhi  Any more issue with the updated [qemu tutorial](https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html) please re-open it?

closed 2021-07-06 by fuzhongl