projectacrn/acrn-hypervisor #5929: ADL platform enabling

lifeix, 2021-04-14:


lifeix, 2021-04-14:
[External_System_ID]ACRN-7009

NanlinXie, 2021-04-27:
Doc Impact Analysis: 

- [ ] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [ ] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

gvancuts, 2021-05-07:
We will need to add this to our list of Supported Hardware: https://projectacrn.github.io/latest/reference/hardware.html

NanlinXie, 2021-06-18:
Suggest we update after we have commercial ADL board in future. @gvancuts, I will remove v2.5 label for this feature.

dbkinder, 2022-01-26:
@NanlinXie What's the target milestone for this?

NanlinXie, 2022-05-04:
Currently we have no schedule plan for document update, close this for now.

closed 2022-05-04 by NanlinXie