projectacrn/acrn-hypervisor #5931: [WindowsGuest] It is not seen and detected Oracle VirtIO SCSI Virtual Disk

ionutnechita, 2021-04-14:
Hi ACRN Team,

I try to install WaaG in ACRN SOS.
But not working. VirtIO Disk is not detected.
I using ACRN 2.4 Stable, but i try with ACRN 2.5 Unstable and this issue is same.

SOS:
OpenSUSE Tumbleweed
ACRN 2.4
Scenario: Industry / Industrial
Oracle ISO for Windows VirtIO Drivers
GOP VBIOS for NUC7I7BNH

WaaG / ISO for install:
Win10_20H2_v2_English_x64.iso

```
NAME="openSUSE Tumbleweed"
VERSION="20210408"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20210408"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20210408"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo"

uname -a
Linux demo-acrn-opensuse-lab1 5.4.106-PKT-200203T060100Z-106355-g3dec93a158e1 #2 SMP PREEMPT Wed Apr 14 14:12:06 EEST 2021 x86_64 x86_64 x86_64 GNU/Linux

acrn-dm --version
DM version is: 2.4-c80508fd7-dirty (daily tag:acrn-2021w14.5-180000p), build by root@2021-04-14 17:08:40

SOS partition: 
/dev/sda2 on / type xfs (rw,relatime,attr2,inode64,logbufs=8,logbsize=32k,noquota)
```

```
#!/bin/bash
# Copyright (C) 2019 Intel Corporation.
# SPDX-License-Identifier: BSD-3-Clause

function launch_win()
{
vm_name=win_vm$1

#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

ip tuntap add tap0 mode tap
ip link set dev tap0 up
brctl addif acrn-br0 tap0
ip link set tap0 master acrn-br0

gpudevice=`cat /sys/bus/pci/devices/0000:00:02.0/device`

echo "8086 $gpudevice" > /sys/bus/pci/drivers/pci-stub/new_id
echo "0000:00:02.0" > /sys/bus/pci/devices/0000:00:02.0/driver/unbind
echo "0000:00:02.0" > /sys/bus/pci/drivers/pci-stub/bind

audiodevice=`cat /sys/bus/pci/devices/0000:00:1f.3/device`

echo "8086 $audiodevice" > /sys/bus/pci/drivers/pci-stub/new_id
echo "0000:00:1f.3" > /sys/bus/pci/devices/0000:00:1f.3/driver/unbind
echo "0000:00:1f.3" > /sys/bus/pci/drivers/pci-stub/bind

#usbdevice=`cat /sys/bus/pci/devices/0000:00:14.0/device`

#echo "8086 $usbdevice" > /sys/bus/pci/drivers/pci-stub/new_id
#echo "0000:00:14.0" > /sys/bus/pci/devices/0000:00:14.0/driver/unbind
#echo "0000:00:14.0" > /sys/bus/pci/drivers/pci-stub/bind

#for memsize setting
mem_size=8192M

acrn-dm -A -m $mem_size -s 0:0,hostbridge -s 1:0,lpc \
  -s 2,passthru,0/2/0,gpu \
  -s 3,virtio-blk,/usr/share/acrn/conf/win10-2020.img \
  -s 4,virtio-net,tap0,mac=98:4f:e1:44:44:44 \
  -s 5,xhci,1-1,1-2 \
  -s 6,passthru,0/1f/3 \
  --ovmf /usr/share/acrn/bios/OVMF_debug.fd.gop \
  -s 7,ahci,cd:/usr/share/acrn/conf/winvirtio.iso \
  -s 8,ahci,cd:/usr/share/acrn/conf/Win10_20H2_v2_English_x64.iso \
  --pm_notify_channel power_button \
  $vm_name
}

#  -s 5,xhci,1-1,1-2 \
#  -s 5,passthru,0/14/0 \
#  -s 7,ahci,cd:/usr/share/acrn/conf/openSUSE-Tumbleweed-DVD-x86_64-Current.iso \

# offline SOS CPUs except BSP before launch UOS
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "$online" = "1" ]; do
                        sleep 1
                        echo 0 > $i/online
                        online=`cat $i/online`
                done
                echo $idx > /sys/class/vhm/acrn_vhm/offline_cpu
        fi
done

launch_win 1 "64 448 8"
```

```
acrn-hypervisor git:
commit c80508fd7101e53790f41f0110e7f7eda74fbac8 (HEAD -> release_2.4, tag: v2.4, origin/release_2.4)
Author: wenlingz <wenling.zhang@intel.com>
Date:   Thu Apr 8 14:56:20 2021 +0800

    version:v2.4

    Signed-off-by: wenlingz <wenling.zhang@intel.com>

commit 234d4c753fdd4cdadceb2be004f41487b9afec07
Author: David B. Kinder <david.b.kinder@intel.com>
Date:   Tue Apr 6 13:50:28 2021 -0700

    doc: update release_2.4 with master content

    Signed-off-by: David B. Kinder <david.b.kinder@intel.com>

    Tracked-On: #5692

commit 3991c48284c11b2e80fb144870b7c86bcd09b89b (tag: acrn-2021w14.5-180000p)
Author: ppsun <peng.p.sun@intel.com>
Date:   Mon Mar 29 08:53:29 2021 +0000

    DM: gvt: Identical mapping for GPU DSM refine to support EHL/TGL

    Windows graphic driver obtains DSM address from in-BAR mmio register
    which has passthroughed. Not like the other platforms obtained from
    pci configure space register which has virtualized. GPU GuC must use
    WOPCM in DSM, besides, Windows OS wants to manage DSM also. These two
    reason force acrn has to keep identical mapping to avoid trap mmio
    BAR to do the emulation.

    Tracked-On: #5880
    Signed-off-by: Peng Sun <peng.p.sun@intel.com>
```

```
acrn-kernel git:
commit 3dec93a158e15b5ceb3d09708d299721193d7a79 (HEAD -> master, origin/master, origin/HEAD)
Author: Yonghua Huang <yonghua.huang@intel.com>
Date:   Mon Mar 22 11:30:17 2021 +0800

    release MSIX resource when unload uio driver

     MSI-x resource should be released when uio_pci_generic
     driver is unloaded.

    Tracked-On: projectacrn/acrn-hypervisor#5889
    Signed-off-by: Yonghua Huang <yonghua.huang@intel.com>

commit eede4a12883adda55305171c5822a5c1059bb249
Author: Geoffroy Van Cutsem <geoffroy.vancutsem@intel.com>
Date:   Fri Feb 19 19:28:19 2021 +0100

    Add Gfx firmware for Apollo Lake (APL)

    Import firmware for the Apollo Lake (APL) integrated graphics.

    Tracked-On: projectacrn/acrn-hypervisor#5752
    Signed-off-by: Geoffroy Van Cutsem <geoffroy.vancutsem@intel.com>

```

ionutnechita, 2021-04-15:
I've tried the --windows parameter now, and it looks like I'm going to install.
So without the --windows parameter, the Windows operating system cannot be installed.

ionutnechita, 2021-04-16:
Hi @gvancuts,

In documentation page for "--windows" parameter.
https://projectacrn.github.io/latest/user-guides/acrn-dm-parameters.html?highlight=windows

--windows
This option is used to run Windows User VMs. It supports Oracle virtio-blk, virtio-net and virtio-input devices for Windows guests with secure boot.

usage:

--windows

A note may be mentioned to say that this parameter is important to detect the disk correctly.
Without it included in acrn-dm, Windows OS cannot be installed or started.

fuzhongl, 2021-05-03:
@ionutnechita Sorry for replying so later!
As mentioned in following link, --windows is MUST parameter  for WaaG.
https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html

closed 2021-05-03 by fuzhongl

gvancuts, 2021-05-04:
FYI, I also filed #5962 recently. It is also related to this.