projectacrn/acrn-hypervisor #5936: Cannot start VM without life_mgnr running on SOS in ACRN 2.4

tttech-industrial-buchsbaum, 2021-04-15:
Up till now we are using ACRN 2.0 on a Debian base system together with libvirt to control the VMs. One of the VMs is started with RT parameters. This is the command that is generated by the libvirt's ACRN hypervisor driver (deduced from libvirtd logs):

```
  /usr/bin/acrn-dm -A --cpu_affinity 2 -m 512M \
    -U d2795438-25d6-11e8-864e-cb7a18b34643 \
    --lapic_pt --virtio_poll 1000000 -s 0:0,hostbridge \
    -s 0:6:0,virtio-blk,/var/lib/libvirt/images/rtvm.raw \
    -s 0:3:0,virtio-net,tap0,mac=xx:xx:xx:xx:xx:xx \
    -s 0:4:0,virtio-net,tap1,mac=yy:yy:yy:yy:yy:yy \
    -s 0:5:0,virtio-net,tap2,mac=zz:zz:zz:zz:zz:zz \
    -s 0:2:0,passthru,1/0/0 \
    -s 0:8:0,virtio-console,@pty:pty_port \
    --pm_notify_channel uart --pm_by_vuart tty,/dev/ttyS1 \
    --logger_setting '"kmsg,level=3"' --ovmf /usr/share/acrn/bios/OVMF.fd rtvm
```

As you can see we use vuart based power mangement channel to the VM where life_mngr is running (`/usr/bin/life_mngr uos /dev/ttyS1`) to detect any shutdwon requests from acrn-dm on SOS.

On ACRN 2.0 using `virsh shutdown rtvm` triggers the following sequence:

On SOS:
* `virsh shutdown rtvm`: make libvirtd (acrn_driver) call `acrnctl stop rtvm`
* in acrn-dm's vm_stop-handler: write 'shutdown' via /dev/ttyS1 SOS vuart

in VM:
* life_mngr listening on /dev/ttyS1 VM vuart receives 'shutdown' and calls `poweroff`

To summarize: **There is no life-mngr required or involved on SOS!**

But when changing to ACRN 2.4, the VM above does not start anymore:
```
root@host:~# /usr/bin/acrn-dm -A --cpu_affinity 2 -m 512M \
    -U d2795438-25d6-11e8-864e-cb7a18b34643 \
    --lapic_pt --virtio_poll 1000000 -s 0:0,hostbridge \
    -s 0:6:0,virtio-blk,/var/lib/libvirt/images/rtvm.raw \
    -s 0:3:0,virtio-net,tap0,mac=xx:xx:xx:xx:xx:xx \
    -s 0:4:0,virtio-net,tap1,mac=yy:yy:yy:yy:yy:yy \
    -s 0:5:0,virtio-net,tap2,mac=zz:zz:zz:zz:zz:zz \
    -s 0:2:0,passthru,1/0/0 \
    -s 0:8:0,virtio-console,@pty:pty_port \
    --pm_notify_channel uart --pm_by_vuart tty,/dev/ttyS1 \
    --logger_setting '"kmsg,level=3"' --ovmf /usr/share/acrn/bios/OVMF.fd rtvm

SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
logger: name="kmsg, level=3
vm_create: rtvm
VHM api version 1.0
vm_setup_memory: size=0x20000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/rtvm/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/rtvm/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:257/257 page size:0x200000
level 1 free/need pages:0/0 page size:0x40000000

try to setup hugepage with:
	level 0 - lowmem 0x20000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7fc8f28e5000 -> baseaddr 0x0x7fc8f2a00000
mmap 0x20000000@0x7fc8f2a00000
touch 256 pages with pagesz 0x200000
mmap 0x200000@0x7fc9f2800000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x20000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
vm_init_vdevs
pm_by_vuart_init idx: 1, path: /dev/ttyS1
initiate a connection on a socket error
create socket to connect life-cycle manager failed
Unable to init vdev (111)
```
Further investigations show, acrn-dm now **REQUIRES** a socket connection to a life_mngr instance running on SOS otherwise pm_vuart setup fails due to the following lines in `start_pm_monitor_thread()` introduced in e40d9d3b6c and e4d2f9a159, respectively, from issue #5736:
```
+       if (pm_setup_socket()) {
+               pr_err("create socket to connect life-cycle manager failed\n");
+               return -1;
+       }
```
Why is running life_mngr on SOS mandatory for ACRN 2.4?
Unfortunately I could not find any documentation on life_mngr and how to use and configure it properly. Maybe I overlooked it?

Since there is no need to run life_mngr on SOS at least for my use case (BTW: when do we need life_mngr on SOS, what are use cases behind that?) and I do not think running unnecessary services on a system is a good thing, shouldn't we handle this situation (i.e. missing life_mngr service on SOS) more gracefully?

Best regards
Helmut

NanlinXie, 2021-04-16:
Could you try to remove --pm_notify_channel uart --pm_by_vuart tty,/dev/ttyS1 if the VM is not intended for lifecycle_mngr?

Best wishes!
Nanlin

    -s ***@***.***:pty_port \

    -s ***@***.***:pty_port \

Further investigations show, acrn-dm now REQUIRES a socket connection to a life_mngr instance running on SOS otherwise pm_vuart setup fails due to the following lines in start_pm_monitor_thread() introduced in e40d9d3<https://github.com/projectacrn/acrn-hypervisor/commit/e40d9d3b6c5922cf94b255fe6840394292575f0c> and e4d2f9a<https://github.com/projectacrn/acrn-hypervisor/commit/e4d2f9a15974e914bd41ab2c0847c65d56d5797f>, respectively, from issue #5736<https://github.com/projectacrn/acrn-hypervisor/issues/5736>:

tttech-industrial-buchsbaum, 2021-04-20:
But --pm_notify_channel uart --pm_by_vuart tty,/dev/ttyS1 defines the connection between acrn-dm and the VM and is needed for the life cycle management of the VM! There is just no reason to run life_mngr running on SOS!

yantingj, 2021-04-20:
Can you try to only remove the option "--pm_notify_channel uart"? 
The use case for life_mngr running on SOS is: System shutdown request from a UoS to SoS. In this case, UoS need "--pm_notify_channel uart" in acrn-dm option, and SoS need life_mngr service. 
In your uese case, you only need system shutdown request from SoS to UoS, thus, "--pm_notify_channel uart" and life_mngr are not needed.

gvancuts, 2021-04-20:
We really need a better document that describes the different options and scenarios for managing power management at the system level. I have tried to reverse-engineer some of it already but it appears rather convoluted to me and the language we use in the various options that are linked to this is not really helping me. Net net, I think I can help improve the doc but I need help understand the whole thing first.

closed 2022-07-05 by fuzhongl