projectacrn/acrn-hypervisor #5938: ACRN v2.4 does not run on QEMU

gvancuts, 2021-04-15:
I'm in the process of trying to update our "Running ACRN on QEMU" document to be based on ACRN v2.4 (the latest release as of today).

Unfortunately, ACRN v2.4 does not start. It throws the following error on the console:
```
Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Multiboot Bootloader: GRUB 2.02-2ubuntu8.21
[0us][cpu=0][(null)][sev=3][seq=1]:vlapic: Basic apicv ops is used!
detect_hardware_support, physical-address width (40) over maximum physical-address width (39)
[0us][cpu=0][(null)][sev=1][seq=2]:PANIC: init_pcpu_pre line: 132

[0us][cpu=0][(null)][sev=1][seq=3]:hardware not support!
```
You can see the steps that I have followed in my draft PR #5937

_Originally posted by @gvancuts in https://github.com/projectacrn/acrn-hypervisor/issues/5928#issuecomment-819154146_

NanlinXie, 2021-04-17:
Two things are ongoing on Qemu guide:
1. Fix the error tag in current document, there's mistake for the ACRN tag, it should be ACRN v2.0.
2. Qemu has regression issue after v2.0, we are ongoing to fix that and refresh the QEMU guide to the latest ACRN.

gvancuts, 2021-04-22:
Additional information on this issue: that error is gone since 628bca5cad3edd7a9f6d1abdbd29fa38964b6239 (more specifically this updated define: `#define MAXIMUM_PA_WIDTH	46U	/* maximum physical-address width */` is what gets us passed the error above).

It does not however work yet, the hypervisor successfully starts the Service VM but it does not boot. There is a kernel crash in the Service VM and then it stops - here is the log:

```
Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Multiboot Bootloader: GRUB 2.02-2ubuntu8.21
[0us][cpu=0][(null)][sev=1][seq=1]:add_pdpte, pdpte 0x100000000 is already present!

[0us][cpu=0][(null)][sev=1][seq=2]:add_pdpte, pdpte 0x140000000 is already present!

ACRN Hypervisor
calibrate_tsc, tsc_khz=2593109
[47267440us][cpu=0][(null)][sev=2][seq=3]:HV version 2.5-unstable-2021-04-22 11:47:32-628bca5ca DBG (daily tag:acrn-2021w17.3-180000p) sdc@qemu build by gvancuts, start time 47254666us
[47269433us][cpu=0][(null)][sev=2][seq=4]:API version 1.0
[47270067us][cpu=0][(null)][sev=2][seq=5]:Detect processor: Intel(R) Core(TM) i7-6770HQ CPU @ 2.60GHz
ACRN:\>[47756889us][cpu=0][(null)][sev=2][seq=30]:Start VM id: 0 name: ACRN SOS VM
[47758096us][cpu=0][vm0:vcpu0][sev=3][seq=31]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[47759903us][cpu=0][vm0:vcpu0][sev=3][seq=32]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[49091613us][cpu=0][vm0:vcpu0][sev=3][seq=35]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[49094250us][cpu=0][vm0:vcpu0][sev=3][seq=36]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[49290811us][cpu=0][vm0:vcpu0][sev=3][seq=37]:vlapic: Start Secondary VCPU1 for VM[0]...
[49292065us][cpu=1][vm0:vcpu1][sev=3][seq=38]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[49293858us][cpu=1][vm0:vcpu1][sev=3][seq=39]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[49373329us][cpu=0][vm0:vcpu0][sev=3][seq=41]:vlapic: Start Secondary VCPU2 for VM[0]...
[49374757us][cpu=2][vm0:vcpu2][sev=3][seq=42]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[49376589us][cpu=2][vm0:vcpu2][sev=3][seq=43]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[49454687us][cpu=0][vm0:vcpu0][sev=3][seq=45]:vlapic: Start Secondary VCPU3 for VM[0]...
[49456201us][cpu=3][vm0:vcpu3][sev=3][seq=46]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[49458134us][cpu=3][vm0:vcpu3][sev=3][seq=47]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[49944941us][cpu=2][vm0:vcpu2][sev=3][seq=49]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[51708011us][cpu=2][vm0:vcpu2][sev=3][seq=50]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[51736477us][cpu=2][vm0:vcpu2][sev=3][seq=51]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[51763639us][cpu=2][vm0:vcpu2][sev=3][seq=52]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[51792105us][cpu=2][vm0:vcpu2][sev=3][seq=53]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[51820258us][cpu=2][vm0:vcpu2][sev=3][seq=54]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[51848596us][cpu=2][vm0:vcpu2][sev=3][seq=55]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[51968034us][cpu=1][vm0:vcpu1][sev=3][seq=56]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[51975924us][cpu=1][vm0:vcpu1][sev=3][seq=57]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
AA[B[B[B

Error: Invalid command.
ACRN:\>
ACRN:\>
ACRN:\>
ACRN:\>
ACRN:\>vm 

Error: Invalid command.
ACRN:\>vm_list

VM_UUID                          VM_ID VM_NAME                          VM_STATE
================================ ===== ================================ ========
dbbbd4347a574216a12c2201f1ab0240   0   ACRN SOS VM                      Running 
ACRN:\>vm_console 0

----- Entering VM 0 Shell -----
i 0000:00:01.3:   bridge window [mem 0xfd600000-0xfd7fffff]
[    1.900344] pci 0000:00:01.3:   bridge window [mem 0xfe400000-0xfe5fffff 64bit pref]
[    1.905334] pci 0000:00:01.4: PCI bridge to [bus 05]
[    1.908507] pci 0000:00:01.4:   bridge window [io  0x5000-0x5fff]
[... 213 lines cut ...]
[  247.047342]  default_idle_call+0x28/0x40
[  247.047343]  do_idle+0x1b7/0x260
[  247.047343]  cpu_startup_entry+0x1d/0x20
[  247.047343]  start_secondary+0x163/0x1b0
[  247.047344]  secondary_startup_64+0xa4/0xb0
[  247.047345] NMI backtrace for cpu 0
[  247.047389] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.106-PKT-200203T060100Z-g3dec93a158e1 #1
[  247.047389] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 1.13.0-1ubuntu1.1 04/01/2014
[  247.047390] RIP: 0010:debug_smp_processor_id+0x0/0xe0
[  247.047391] Code: af ff 4c 89 6d c8 e9 a4 fe ff ff 48 c7 00 00 00 00 00 c7 40 08 00 00 00 00 4c 89 7d c8 e9 8d fe ff ff 90 90 90 90 90 90 90 90 <55> 48 89 e5 41 54 53 65 8b 1d 5a 09 a4 4e 65 8b 05 43 
63 a4 4e a9
[  247.047391] RSP: 0000:ffffffffb2c03e00 EFLAGS: 00000296
[  247.047392] RAX: ffffffffb2c03e28 RBX: 0000000000000000 RCX: 0000000000000000
[  247.047393] RDX: 0000000000000001 RSI: ffffffffb102a705 RDI: ffffffffb1e7239f
[  247.047393] RBP: ffffffffb2c03e28 R08: 0000000000000000 R09: 000000560fc8b800
[  247.047394] R10: ffffffffb2c03dd0 R11: 0000000000000000 R12: ffffffffb2dd3670
[  247.047394] R13: 0000000000000000 R14: 0000000000000000 R15: ffffffffb2ef12e0
[  247.047395] FS:  0000000000000000(0000) GS:ffff944c39000000(0000) knlGS:0000000000000000
[  247.047395] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  247.047396] CR2: 0000000000000000 CR3: 000000003fc13001 CR4: 0000000000360ef0
[  247.047396] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  247.047397] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  247.047397] Call Trace:
[  247.047398]  ? default_idle+0x32/0x190
```

gvancuts, 2021-05-12:
I turned on more debugging options in the hypervisor (v2.4 + the diff listed at the bottom [1])... here is the new trace:
```
Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Multiboot Bootloader: GRUB 2.02-2ubuntu8.21
[0us][cpu=0][(null)][sev=3][seq=1]:vlapic: Basic apicv ops is used!
[0us][cpu=0][(null)][sev=6][seq=2]:mmap addr 0x11200080 entries 10

[0us][cpu=0][(null)][sev=6][seq=3]:mmap table: 0 type: 0x1
[0us][cpu=0][(null)][sev=6][seq=4]:Base: 0x0000000000000000 length: 0x000000000009fc00

[0us][cpu=0][(null)][sev=6][seq=5]:mmap table: 1 type: 0x2
[0us][cpu=0][(null)][sev=6][seq=6]:Base: 0x000000000009fc00 length: 0x0000000000000400

[0us][cpu=0][(null)][sev=6][seq=7]:mmap table: 2 type: 0x2
[0us][cpu=0][(null)][sev=6][seq=8]:Base: 0x00000000000f0000 length: 0x0000000000010000

[0us][cpu=0][(null)][sev=6][seq=9]:mmap table: 3 type: 0x1
[0us][cpu=0][(null)][sev=6][seq=10]:Base: 0x0000000000100000 length: 0x000000007fed8000

[0us][cpu=0][(null)][sev=6][seq=11]:mmap table: 4 type: 0x2
[0us][cpu=0][(null)][sev=6][seq=12]:Base: 0x000000007ffd8000 length: 0x0000000000028000

[0us][cpu=0][(null)][sev=6][seq=13]:mmap table: 5 type: 0x2
[0us][cpu=0][(null)][sev=6][seq=14]:Base: 0x00000000b0000000 length: 0x0000000010000000

[0us][cpu=0][(null)][sev=6][seq=15]:mmap table: 6 type: 0x2
[0us][cpu=0][(null)][sev=6][seq=16]:Base: 0x00000000fed1c000 length: 0x0000000000004000

[0us][cpu=0][(null)][sev=6][seq=17]:mmap table: 7 type: 0x2
[0us][cpu=0][(null)][sev=6][seq=18]:Base: 0x00000000feffc000 length: 0x0000000000004000

[0us][cpu=0][(null)][sev=6][seq=19]:mmap table: 8 type: 0x2
[0us][cpu=0][(null)][sev=6][seq=20]:Base: 0x00000000fffc0000 length: 0x0000000000040000

[0us][cpu=0][(null)][sev=6][seq=21]:mmap table: 9 type: 0x1
[0us][cpu=0][(null)][sev=6][seq=22]:Base: 0x0000000100000000 length: 0x0000000080000000

[0us][cpu=0][(null)][sev=6][seq=23]:HV MMU Initialization
[0us][cpu=0][(null)][sev=6][seq=24]:mmu_add, paddr 0x0, vaddr 0x0, size 0x180000000

[0us][cpu=0][(null)][sev=6][seq=25]:add_pdpte, paddr: 0x0, vaddr: [0x0 - 0x180000000]

[0us][cpu=0][(null)][sev=6][seq=26]:mmu_modify_or_del, vaddr: 0x0, size: 0x80000000

[0us][cpu=0][(null)][sev=6][seq=27]:modify_or_del_pdpte, vaddr: [0x0 - 0x80000000]

[0us][cpu=0][(null)][sev=6][seq=28]:mmu_modify_or_del, vaddr: 0x100000000, size: 0x80000000

[0us][cpu=0][(null)][sev=6][seq=29]:modify_or_del_pdpte, vaddr: [0x100000000 - 0x180000000]

[0us][cpu=0][(null)][sev=6][seq=30]:mmu_modify_or_del, vaddr: 0x11000000, size: 0x7800000

[0us][cpu=0][(null)][sev=6][seq=31]:modify_or_del_pdpte, vaddr: [0x11000000 - 0x18800000]

[0us][cpu=0][(null)][sev=6][seq=32]:split_large_page, paddr: 0x0, pbase: 0x12406000

[0us][cpu=0][(null)][sev=6][seq=33]:modify_or_del_pde, vaddr: [0x11000000 - 0x18800000]

[0us][cpu=0][(null)][sev=6][seq=34]:mmu_modify_or_del, vaddr: 0x11000000, size: 0x200000

[0us][cpu=0][(null)][sev=6][seq=35]:modify_or_del_pdpte, vaddr: [0x11000000 - 0x11200000]

[0us][cpu=0][(null)][sev=6][seq=36]:modify_or_del_pde, vaddr: [0x11000000 - 0x11200000]

[0us][cpu=0][(null)][sev=6][seq=37]:mmu_modify_or_del, vaddr: 0x11400000, size: 0x1000000

[0us][cpu=0][(null)][sev=6][seq=38]:modify_or_del_pdpte, vaddr: [0x11400000 - 0x12400000]

[0us][cpu=0][(null)][sev=6][seq=39]:modify_or_del_pde, vaddr: [0x11400000 - 0x12400000]
[... 324 lines cut ...]
 /* size of the low MMIO address space: 2GB */
 #define PLATFORM_LO_MMIO_SIZE  0x80000000UL
diff --git a/misc/config_tools/data/qemu/sdc.xml b/misc/config_tools/data/qemu/sdc.xml
index a5584e61f..2bbfc076c 100644
--- a/misc/config_tools/data/qemu/sdc.xml
+++ b/misc/config_tools/data/qemu/sdc.xml
@@ -6,7 +6,7 @@
         <SERIAL_CONSOLE>/dev/ttyS0</SERIAL_CONSOLE>
         <MEM_LOGLEVEL>5</MEM_LOGLEVEL>
         <NPK_LOGLEVEL>5</NPK_LOGLEVEL>
-        <CONSOLE_LOGLEVEL>3</CONSOLE_LOGLEVEL>
+        <CONSOLE_LOGLEVEL>6</CONSOLE_LOGLEVEL>
         <LOG_DESTINATION>7</LOG_DESTINATION>
         <LOG_BUF_SIZE>0x40000</LOG_BUF_SIZE>
     </DEBUG_OPTIONS>
@@ -104,7 +104,7 @@
         <rootfs>/dev/vda1</rootfs>
         <bootargs>
         earlyprintk=serial,ttyS0,115200n8 rw rootwait console=tty0 consoleblank=0 no_timer_check ignore_loglevel
-        ignore_loglevel no_timer_check intel_iommu=off tsc=reliable hvlog=2M@0x1FE00000
+        intel_iommu=off tsc=reliable hvlog=2M@0x1FE00000
         </bootargs>
     </board_private>
   </vm>
```

fuzhongl, 2021-05-13:
@gvancuts Please boot into native and check if `intel_iommu=on` is in your HOST OS kernel cmdline.
If Yes, please remove it and try ACRN on qemu again.

gvancuts, 2021-05-13:
What do you mean by "boot into native"? I have a native Ubuntu 20.04 running (no `intel_iommu` option set via the kernel command-line). I boot QEMU on that native Ubuntu 20.04 and have the choice to run native Ubuntu 18.04 or ACRN + Ubuntu 18.04 in that VM. Neither of those options in the QEMU VM set the `intel_iommu` option on my system.

fuzhongl, 2021-05-13:
@gvancuts 
With following change on v2.4:
MAXIMUM_PA_WIDTH from 39U to 40U
It works well on my Skylake: `[129496563us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel(R) Core(TM) i7-6770HQ CPU @ 2.60GHz`

```
                             GNU GRUB  version 2.02

 +----------------------------------------------------------------------------+
 |setparams 'ACRN hypervisor'                                                 |
 |                                                                            |
 |   gfxmode $linux_gfx_mode                                                  |
 |   insmod gzio                                                              |
 |   insmod part_msdos                                                        |
 |   insmod ext2                                                              |
 |                                                                            |
 |   echo 'Loading ACRN hypervisor with SDC scenario ...'                     |
 |   multiboot --quirk-modules-after-kernel /boot/acrn/acrn.32.out.24_40      |
 |   module /boot/acrn/sos_bzImage Linux_bzImage                              |
 |                                                                            |
 |                                                                            |
 +----------------------------------------------------------------------------+

      Minimum Emacs-like screen editing is supported. TAB lists
      completions. Press Ctrl-x or F10 to boot, Ctrl-c or F2 for
      a command-line or ESC to discard edits and return to the GRUB menu.
  Booting a command list

Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Multiboot Bootloader: GRUB 2.02-2ubuntu8.17
[0us][cpu=0][(null)][sev=3][seq=1]:vlapic: Basic apicv ops is used!
ACRN Hypervisor
calibrate_tsc, tsc_khz=2595532
[129493438us][cpu=0][(null)][sev=2][seq=2]:HV version 2.4-2021-05-08 21:49:28-ade36db37-dirty DBG (daily tag:acrn-2021w14.5-180000p) sdc@qemu build by root, start time 129481439us
[129495780us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[129496563us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel(R) Core(TM) i7-6770HQ CPU @ 2.60GHz
ACRN:\>[130097587us][cpu=0][(null)][sev=2][seq=29]:Start VM id: 0 name: ACRN SOS VM
[130098638us][cpu=0][vm0:vcpu0][sev=3][seq=30]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[130100366us][cpu=0][vm0:vcpu0][sev=3][seq=31]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[132061612us][cpu=0][vm0:vcpu0][sev=3][seq=34]:vlapic: Start Secondary VCPU1 for VM[0]...
[132082236us][cpu=1][vm0:vcpu1][sev=3][seq=35]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[132084214us][cpu=1][vm0:vcpu1][sev=3][seq=36]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[132182246us][cpu=0][vm0:vcpu0][sev=3][seq=38]:vlapic: Start Secondary VCPU2 for VM[0]...
[132202858us][cpu=2][vm0:vcpu2][sev=3][seq=39]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[132204739us][cpu=2][vm0:vcpu2][sev=3][seq=40]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[132301329us][cpu=0][vm0:vcpu0][sev=3][seq=42]:vlapic: Start Secondary VCPU3 for VM[0]...
[132321919us][cpu=3][vm0:vcpu3][sev=3][seq=43]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[132323585us][cpu=3][vm0:vcpu3][sev=3][seq=44]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

ACRN:\>vcpu_list

VM ID    PCPU ID    VCPU ID    VCPU ROLE    VCPU STATE    THREAD STATE
=====    =======    =======    =========    ==========    ==========
  0         0          0       PRIMARY      Running          BLOCKED
  0         1          1       SECONDARY    Running          BLOCKED
  0         2          2       SECONDARY    Running          BLOCKED
  0         3          3       SECONDARY    Running          BLOCKED
ACRN:\>vm_console 0

----- Entering VM 0 Shell -----
R: enabled with IRQ 28
pcieport 0000:00:01.5: PME: Signaling with IRQ 29
[... 169 lines cut ...]
systemd[1]: Listening on Syslog Socket.
EXT4-fs (vda1): re-mounted. Opts: errors=remount-ro
Adding 1557568k swap on /swapfile.  Priority:-2 extents:5 across:1590336k
systemd-journald[169]: Received request to flush runtime journal from PID 1
virtio_net virtio0 enp1s0: renamed from eth0

Ubuntu 18.04.5 LTS ubuntu ttyS0

ubuntu login: root
Password:
Last login: Thu May 13 18:13:38 PDT 2021 on ttyS0
Welcome to Ubuntu 18.04.5 LTS (GNU/Linux 5.4.51-acrn-kernel-sos x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

 * Canonical Livepatch is available for installation.
   - Reduce system reboots and improve kernel security. Activate at:
     https://ubuntu.com/livepatch
New release '20.04.2 LTS' available.
Run 'do-release-upgrade' to upgrade to it.

root@ubuntu:~#
```

fuzhongl, 2021-05-13:
> What do you mean by "boot into native"? I have a native Ubuntu 20.04 running (no `intel_iommu` option set via the kernel command-line). I boot QEMU on that native Ubuntu 20.04 and have the choice to [...]
 @gvancuts  
 "boot into native" means bare metal OS; no related with qemu.
I can reproduce your issue if adding `intel_iommu=on` in bare metal OS
Could you please share `cat /proc/cmdline` for the bare metal OS ?

gvancuts, 2021-05-17:
Hi @fuzhongl ,

Here are the details:
```
$ cat /proc/cmdline 
BOOT_IMAGE=/boot/vmlinuz-5.8.0-49-generic root=UUID=836b6013-a0ef-4fe4-a428-22a0f2d5125f ro quiet splash vt.handoff=7
```

IOMMU relevant bits from the kernel config (full config attached -
[config-5.8.0-49-generic.txt](https://github.com/projectacrn/acrn-hypervisor/files/6492273/config-5.8.0-49-generic.txt):
```
$ grep IOMMU /boot/config-5.8.0-49-generic
CONFIG_IRQ_MSI_IOMMU=y
CONFIG_GART_IOMMU=y
CONFIG_VFIO_IOMMU_TYPE1=y
CONFIG_VFIO_NOIOMMU=y
CONFIG_IOMMU_IOVA=y
CONFIG_IOMMU_API=y
CONFIG_IOMMU_SUPPORT=y
# Generic IOMMU Pagetable Support
# end of Generic IOMMU Pagetable Support
# CONFIG_IOMMU_DEBUGFS is not set
# CONFIG_IOMMU_DEFAULT_PASSTHROUGH is not set
CONFIG_IOMMU_DMA=y
CONFIG_AMD_IOMMU=y
CONFIG_AMD_IOMMU_V2=m
CONFIG_INTEL_IOMMU=y
CONFIG_INTEL_IOMMU_SVM=y
# CONFIG_INTEL_IOMMU_DEFAULT_ON is not set
CONFIG_INTEL_IOMMU_FLOPPY_WA=y
# CONFIG_INTEL_IOMMU_SCALABLE_MODE_DEFAULT_ON is not set
CONFIG_HYPERV_IOMMU=y
CONFIG_IOMMU_HELPER=y
# CONFIG_IOMMU_DEBUG is not set
```

gvancuts, 2021-05-17:
Hi @fuzhongl , I have just tried with `intel_iommu=off` but I am still seeing this issue.
* Host OS:
```
$ cat /proc/cmdline 
BOOT_IMAGE=/boot/vmlinuz-5.8.0-53-generic root=UUID=836b6013-a0ef-4fe4-a428-22a0f2d5125f ro quiet splash vt.handoff=7 intel_iommu=off
```

* Hypervisor console:
```
Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Multiboot Bootloader: GRUB 2.02-2ubuntu8.21
[0us][cpu=0][(null)][sev=3][seq=1]:vlapic: Basic apicv ops is used!
ACRN Hypervisor
calibrate_tsc, tsc_khz=2592081
[4123834us][cpu=0][(null)][sev=2][seq=2]:HV version 2.4-2021-05-12 17:00:16-c80508fd7-dirty DBG (daily tag:acrn-2021w14.5-180000p) sdc@qemu build by acrn, start time 4111445us
[4126690us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[4127505us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel(R) Core(TM) i7-6770HQ CPU @ 2.60GHz
ACRN:\>[4594267us][cpu=0][(null)][sev=2][seq=29]:Start VM id: 0 name: ACRN SOS VM
[4595442us][cpu=0][vm0:vcpu0][sev=3][seq=30]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[4597150us][cpu=0][vm0:vcpu0][sev=3][seq=31]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[5810770us][cpu=0][vm0:vcpu0][sev=3][seq=34]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[5813182us][cpu=0][vm0:vcpu0][sev=3][seq=35]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[5950190us][cpu=0][vm0:vcpu0][sev=3][seq=36]:vlapic: Start Secondary VCPU1 for VM[0]...
[5952028us][cpu=1][vm0:vcpu1][sev=3][seq=37]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[5953863us][cpu=1][vm0:vcpu1][sev=3][seq=38]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[6033772us][cpu=0][vm0:vcpu0][sev=3][seq=40]:vlapic: Start Secondary VCPU2 for VM[0]...
[6035372us][cpu=2][vm0:vcpu2][sev=3][seq=41]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[6037820us][cpu=2][vm0:vcpu2][sev=3][seq=42]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[6117158us][cpu=0][vm0:vcpu0][sev=3][seq=44]:vlapic: Start Secondary VCPU3 for VM[0]...
[6118602us][cpu=3][vm0:vcpu3][sev=3][seq=45]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[6120474us][cpu=3][vm0:vcpu3][sev=3][seq=46]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[6604260us][cpu=0][vm0:vcpu0][sev=3][seq=48]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[8474796us][cpu=0][vm0:vcpu0][sev=3][seq=49]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[8500459us][cpu=0][vm0:vcpu0][sev=3][seq=50]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[8525535us][cpu=0][vm0:vcpu0][sev=3][seq=51]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[8550823us][cpu=0][vm0:vcpu0][sev=3][seq=52]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[8576466us][cpu=0][vm0:vcpu0][sev=3][seq=53]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[8600786us][cpu=0][vm0:vcpu0][sev=3][seq=54]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[8720379us][cpu=2][vm0:vcpu2][sev=3][seq=55]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
[8727472us][cpu=2][vm0:vcpu2][sev=3][seq=56]:iommu: DMAR OP Timeout! @ dmar_issue_qi_request
```

* Service VM console output
```
----- Entering VM 0 Shell -----
i 0000:00:01.3:   bridge window [mem 0xfd600000-0xfd7fffff]
[    1.551532] pci 0000:00:01.3:   bridge window [mem 0xfe400000-0xfe5fffff 64bit pref]
[    1.556169] pci 0000:00:01.4: PCI bridge to [bus 05]
[    1.559123] pci 0000:00:01.4:   bridge window [io  0x5000-0x5fff]
[    1.562721] pci 0000:00:01.4:   bridge window [mem 0xfd400000-0xfd5fffff]
[    1.566667] pci 0000:00:01.4:   bridge window [mem 0xfe200000-0xfe3fffff 64bit pref]
[    1.571259] pci 0000:00:01.5: PCI bridge to [bus 06]
[    1.574185] pci 0000:00:01.5:   bridge window [io  0x6000-0x6fff]
[    1.577939] pci 0000:00:01.5:   bridge window [mem 0xfd200000-0xfd3fffff]
[    1.581930] pci 0000:00:01.5:   bridge window [mem 0xfe000000-0xfe1fffff 64bit pref]
[    1.586455] pci_bus 0000:00: resource 4 [io  0x0000-0x0cf7 window]
[    1.589989] pci_bus 0000:00: resource 5 [io  0x0d00-0xffff window]
[    1.593570] pci_bus 0000:00: resource 6 [mem 0x000a0000-0x000bffff window]
[    1.597598] pci_bus 0000:00: resource 7 [mem 0x80000000-0xafffffff window]
[    1.601502] pci_bus 0000:00: resource 8 [mem 0xc0000000-0xfebfffff window]
[    1.605480] pci_bus 0000:00: resource 9 [mem 0x180000000-0x97fffffff window]
[... 244 lines cut ...]
[269956416us][cpu=0][(null)][sev=2][seq=35]:addr(0x1120ac70)	0x00000000110548b0  0x00000000696e746c  0x0000000000000000  0x0000000000000000

[269959587us][cpu=0][(null)][sev=2][seq=36]:addr(0x1120ac90)	0x0000000100000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[269962778us][cpu=0][(null)][sev=2][seq=37]:addr(0x1120acb0)	0x000000001120acb0  0x000000001120acb0  0x0000000000000000  0x0000000000000000

[269965873us][cpu=0][(null)][sev=2][seq=38]:

[269966862us][cpu=0][(null)][sev=2][seq=39]:Host Call Trace:

[269968064us][cpu=0][(null)][sev=2][seq=40]:----> 0x000000001100c966

[269970155us][cpu=0][(null)][sev=2][seq=41]:----> 0x000000001100c352

[269972316us][cpu=0][(null)][sev=2][seq=42]:----> 0x000000001101645e

[269973785us][cpu=0][(null)][sev=2][seq=43]:----> 0x000000001101700c

[269975176us][cpu=0][(null)][sev=2][seq=44]:----> 0x0000000011017118

[269976694us][cpu=0][(null)][sev=2][seq=45]:----> 0x00000000110081e8

[269978283us][cpu=0][(null)][sev=2][seq=46]:

```

fuzhongl, 2021-05-17:
@gvancuts 
Thanks for update!
Could you please add `memmap=0x200000$0x1FE00000` in grub menu?
`multiboot --quirk-modules-after-kernel /boot/acrn/acrn.32.out memmap=0x200000\$0x1FE00000`
If still fail, please share the BIOS version of your Skylake board.

gvancuts, 2021-05-17:
> Could you please add `memmap=0x200000$0x1FE00000` in grub menu?
> `multiboot --quirk-modules-after-kernel /boot/acrn/acrn.32.out memmap=0x200000\$0x1FE00000`

Do I need the backslash in there?

> If still fail, please share the BIOS version of your Skylake board.

Should I test this on v2.4 or the latest version of ACRN (or both)?

I will add my BIOS information later!

Thanks

gvancuts, 2021-05-17:
Bios information:
```
BIOS Information
	Vendor: Intel Corp.
	Version: KYSKLi70.86A.0050.2017.0831.1924
	Release Date: 08/31/2017
```
There is a much more recent bios available. I will update it and try again. The `memmap` argument did not make any difference (using a v2.4 ACRN hypervisor).

gvancuts, 2021-05-17:
Latest test setup - unfortunately with the same results.

Bios (latest)
```
BIOS Information
        Vendor: Intel Corp.
        Version: KYSKLi70.86A.0071.2020.0909.1612
        Release Date: 09/09/2020
```
Host kernel command-line:
```
$ cat /proc/cmdline 
BOOT_IMAGE=/boot/vmlinuz-5.8.0-53-generic root=UUID=836b6013-a0ef-4fe4-a428-22a0f2d5125f ro quiet splash vt.handoff=7 intel_iommu=off
```

Using ACRN v2.4 with patch:
```
acrn@acrnsos:~/acrn-hypervisor$ git describe
v2.4
acrn@acrnsos:~/acrn-hypervisor$ git diff
diff --git a/hypervisor/include/arch/x86/page.h b/hypervisor/include/arch/x86/page.h
index 1094fa30d..e7c042903 100644
--- a/hypervisor/include/arch/x86/page.h
+++ b/hypervisor/include/arch/x86/page.h
@@ -14,7 +14,7 @@
 #define PAGE_SIZE      (1U << PAGE_SHIFT)
 #define PAGE_MASK      0xFFFFFFFFFFFFF000UL
 
-#define MAXIMUM_PA_WIDTH       39U     /* maximum physical-address width */
+#define MAXIMUM_PA_WIDTH       40U     /* maximum physical-address width */
 
 /* size of the low MMIO address space: 2GB */
 #define PLATFORM_LO_MMIO_SIZE  0x80000000UL
diff --git a/misc/config_tools/data/qemu/sdc.xml b/misc/config_tools/data/qemu/sdc.xml
index a5584e61f..53333bbdd 100644
--- a/misc/config_tools/data/qemu/sdc.xml
+++ b/misc/config_tools/data/qemu/sdc.xml
@@ -6,7 +6,7 @@
         <SERIAL_CONSOLE>/dev/ttyS0</SERIAL_CONSOLE>
         <MEM_LOGLEVEL>5</MEM_LOGLEVEL>
         <NPK_LOGLEVEL>5</NPK_LOGLEVEL>
-        <CONSOLE_LOGLEVEL>3</CONSOLE_LOGLEVEL>
+        <CONSOLE_LOGLEVEL>5</CONSOLE_LOGLEVEL>
         <LOG_DESTINATION>7</LOG_DESTINATION>
         <LOG_BUF_SIZE>0x40000</LOG_BUF_SIZE>
     </DEBUG_OPTIONS>
@@ -104,7 +104,7 @@
         <rootfs>/dev/vda1</rootfs>
         <bootargs>
         earlyprintk=serial,ttyS0,115200n8 rw rootwait console=tty0 consoleblank=0 no_timer_check ignore_loglevel
-        ignore_loglevel no_timer_check intel_iommu=off tsc=reliable hvlog=2M@0x1FE00000
+        intel_iommu=off tsc=reliable hvlog=2M@0x1FE00000
         </bootargs>
     </board_private>
   </vm>
```
Grub in QEMU VM:
```
$ cat /etc/grub.d/40_custom 
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.

menuentry 'ACRN hypervisor' --class ubuntu --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-simple-e23c76ae-b06d-4a6e-ad42-46b8eedfd7d3' {
   recordfail
   load_video
   gfxmode $linux_gfx_mode
   insmod gzio
   insmod part_msdos
   insmod ext2

   echo 'Loading ACRN hypervisor with SDC scenario ...'
   multiboot --quirk-modules-after-kernel /boot/acrn.32.out memmap=0x200000\$0x1FE00000
   module /boot/bzImage Linux_bzImage
}
```

fuzhongl, 2021-06-04:
narrow down and root caused: setup issue about sos_bzImage.

closed 2021-06-04 by fuzhongl

gvancuts, 2021-06-04:
> narrow down and root caused: setup issue about sos_bzImage.

A new tutorial will be published with the updated instructions (see #5937). The key with regards to the Service VM Linux kernel is to **not** enable the Intel IOMMU config option anymore.