projectacrn/acrn-hypervisor #5939: How to write a application to test Inter-VM Communication

SIMING29, 2021-04-16:
[image]

As shown in the picture, Is there a application demo under linux and windows？
Or where can I find the relevant information? like the ivshmem user-space driver and the API

gvancuts, 2021-04-19:
Hi @zhangshuai75 , this is a gap, we do not have such tutorial or example code. It would be good to have one. @NanlinXie , who can we assign this to?

yonghuah, 2021-04-19:
@zhangshuai75 @gvancuts ,  ACRN implements IVHSHMEM device following below device specification defined by KVM/QEMU community: 
https://git.qemu.org/?p=qemu.git;a=blob_plain;f=docs/specs/ivshmem-spec.txt;hb=HEAD

For usage information, please refer to "Enable Inter-VM Communication Based on Ivshmem“
https://projectacrn.github.io/latest/tutorials/enable_ivshmem.html

-Yonghua

gvancuts, 2021-04-19:
Thanks @yonghuah , I am aware of these documents (and I guess @zhangshuai75 is too), but it would be good to have a small sample app that demonstrates this and can also help users validate that the shared memory segments they set up are working correctly. `man shm_open` (https://man7.org/linux/man-pages/man3/shm_open.3.html) gives a simple example and something similar, but for ACRN would be good to have.

zhangshuai75, 2021-04-20:
@gvancuts @yonghuah , Thanks, our main purpose now is to test the functionality and performance of ACRN, the application development will be put after the verification. From our point of view, it’s better to have a demo, otherwise we may take a lot of time to write and the test results are not comprehensive

gvancuts, 2021-04-20:
We need a tutorial on how to use `ivshmem`, let's track this in this request, re-opening. Please assign someone to this task and @dbkinder @fitchbe and I will help review this.

NanlinXie, 2021-09-03:
@gvancuts, do you agree that we provice some sample code in the document for this requirement?

gvancuts, 2021-09-03:
> @gvancuts, do you agree that we provice some sample code in the document for this requirement?

That works for me as long as it's clear and accurate enough so that there is no ambiguity.

NanlinXie, 2022-01-04:
https://github.com/projectacrn/acrn-hypervisor/pull/6973

closed 2022-07-05 by fuzhongl