projectacrn/acrn-hypervisor #5950: [DX_task]Remove is_rt_vm

NanlinXie, 2021-04-22:
GUEST_FLAG_RT is actually a policy level flag, which should only be aware by up-level, and should not be seen in hypervisor source code.

In the hypervisor, it should not know if a VM is a RT VM, but just know if a VM is using LAPIC pass-thru, and if a VM is a high severity VM.

NanlinXie, 2021-04-22:
[External_System_ID] ACRN-6701

NanlinXie, 2021-04-22:
Doc Impact Analysis: 

- [ ] New feature -> new HLD and usage guide
- [ ] [HLDs](https://projectacrn.github.io/2.3/developer-guides/hld/index.html)
- [ ] [Configuration data](https://projectacrn.github.io/2.3/tutorials/acrn_configuration_tool.html)
- [ ] [Kernel module parameters](https://projectacrn.github.io/2.3/user-guides/kernel-parameters.html)
- [ ] [Device Model (acrn-dm) parameters](https://projectacrn.github.io/2.3/user-guides/acrn-dm-parameters.html)
- [ ] [Hypervisor shell commands](https://projectacrn.github.io/2.3/user-guides/acrn-shell.html)
- [ ] [ACRN tools (debug and/or service)](https://projectacrn.github.io/2.3/develop.html#configuration-and-tools)
- [ ] [Build system](https://projectacrn.github.io/latest/getting-started/building-from-source.html)
- [x] [Other documents (e.g. tutorials)](https://projectacrn.github.io/2.3/develop.html#enable-acrn-features)

gvancuts, 2021-04-22:
Why is this marked as a `DX_task`? I thought it was more of a code clean-up initiative but with in practice little to no effect to the user, or am I missing something?

NanlinXie, 2021-04-27:
Delete GUEST_FLAG_RT and add GUEST_FLAG_FORCE_4K_IPAGE/GUEST_FLAG_NO_NATIVE_WBINVD

NanlinXie, 2021-05-24:
Postponed to future release due to arch open.

closed 2022-03-04 by fuzhongl