projectacrn/acrn-hypervisor #5960: CPU assignment in ACRN

gvancuts, 2021-04-22:
The `--cpu_affinity` parameter (`acrn-dm`) should work within the boundaries of what’s defined in the scenario XML. A meaningful error should be thrown if the launch script uses a CPU assignment that conflicts with that.

gvancuts, 2021-04-22:
> 
> 
> The `--cpu_affinity` parameter (`acrn-dm`) should work within the boundaries of what’s defined in the scenario XML. A meaningful error should be thrown if the launch script uses a CPU assignment that conflicts with that.

A case could also be made to remove the option altogether although I personnally see _some_ value in being able to tweak a little the basic CPU assignment from the scenario file directly from the launch script.

liuhang-bit, 2021-08-11:
@gvancuts 

> A case could also be made to remove the option altogether although I personnally see _some_ value in being able to tweak a little the basic CPU assignment from the scenario file directly from the [...]

@gvancuts  could you elaborate on the case you mentioned above, do you suggest to remove acrn-dm --cpu_affinity ,since it is defined in the vm_config statically.

liuhang-bit, 2021-08-19:
In our existing implementation in ACRN , user set cpu affinity config in scenario xml with config tool UI, for example on chip ,there is 10 pcpu , user setting 1~5 cpu-affinity means 1~5 pcpu for specific post-launched VM running on. then user will use acrn-dm to launch the vm in question, they may specific a pcpu out of the scope 1~5 with --cpu-affinity parameter, which will lead an error reported like this  "Post-launched VM chooses invalid pCPUs" in hypervisor, they could only specify 1~5 pcpu ,or a subset of it .@gvancuts

liuhang-bit, 2021-08-19:
we wiil check the acrn-dm return error for any mismatching

NanlinXie, 2022-01-04:
Will cover in DX debug improvement task

dbkinder, 2022-03-01:
@NanlinXie @liuhang-bit @gvancuts 
It looks like acrn-dm does return an error, but isn't descriptive enough to figure out what happened:
```
failed to create VM POST_STD_VM2, Invalid argument.
```