projectacrn/acrn-hypervisor #5962: ACRN Device Model `--windows` parameter

gvancuts, 2021-04-22:
The `--windows` parameter is used to enable the Oracle VirtIO block device instead of the standard `virtio-blk`.

This option should probably be renamed into something more meaningful, or possibly even as a sub-option of the `virtio-blk` PCI device.

I assume other User VM OSs may want to use that device, not just Windows (or is this really something that can and should only be used with Windows?).

gvancuts, 2021-05-10:
> @gvancuts , could we close this since your patch already been merged?

I think we should still consider renaming this option, or even better, having it as a parameter to the `virtio-blk` PCI device option.

NanlinXie, 2021-07-30:
After discussion, We need add more description to explanation in windows document.
https://projectacrn.github.io/latest/tutorials/waag-secure-boot.html
https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html