projectacrn/acrn-hypervisor #5966: Remove MAX_KATA_VM_NUM variable

gvancuts, 2021-04-23:
There is a hypervisor variable that defines the max number of Kata Containers that can be run on ACRN: https://projectacrn.github.io/latest/reference/config-options.html?highlight=kata_vm#cmdoption-arg-hv.CAPACITIES.MAX_KATA_VM_NUM.

My analysis after a little digging is that we should be able to remove this altogether and treat Kata Containers are normal post-launched User VMs. There is nothing in the code that I saw that was *really* specific to Kata Containers. The couple of things that ought to be fixed are:
* UUID handling: there is one dedicated UUID used for Kata Containers today (see [here](https://github.com/projectacrn/acrn-hypervisor/blob/0305640a5b4e5c039dbace6b8a3570f7329e74fc/misc/config_tools/library/scenario_cfg_lib.py#L55)). We should remove that.
* The fact there is only one UUID defined is actually a problem in itself because that means that any value other than 0 or 1 for MAX_KATA_VM_NUM will not work.
* The ACRN implementation in Kata Containers already defines 8 UUID that can be used (see [here](https://github.com/kata-containers/kata-containers/blob/1c222c75ac046ed0105fac1e00f7defcbd35bad7/src/runtime/virtcontainers/acrn.go#L42-L71)). These should all be known to ACRN (they are not) and there should be some simple logic to use the first free UUID when starting a Kata Container on ACRN

NanlinXie, 2022-01-04:
MAX_KATA_VM_NUM was already removed.

NanlinXie, 2022-01-04:
@gvancuts , do you agree to close it?

closed 2022-01-04 by NanlinXie