projectacrn/acrn-hypervisor #5970: ACRN panic when boot acrn-hypervisor in MAXTANG WHL WL10 board

iaGuoZhi, 2021-04-24:
**Describe the bug**

When I follow ACRN document [Getting Started Guide for ACRN Industry Scenario with Ubuntu Service VM](https://projectacrn.github.io/2.0/getting-started/rt_industry_ubuntu.html) with ACRN v2.0 using MAXTANG WHL WL10 board. ACRN fail to boot

**Platform**

cpu: i5: Intel(R) Core(TM) i5-8265U
service vm: ubuntu20
grub: grub2.04

**Codebase**
I've tried ACRN v2.0 v2.2 v2.4.

**Scenario**

Industry

**To Reproduce**

Steps to reproduce the behavior:

1. install service vm ubuntu20
2. compile acrn hypervisor using `make all BOARD_FILE=misc/acrn-config/xmls/board-xmls/whl-ipc-i5.xml SCENARIO_FILE=misc/acrn-config/xmls/config-xmls/whl-ipc-i5/industry.xml RELEASE=0`, this step produce no error or warning.
3. compile kernel without error.
4. update grub
5. reboot
6. see following error
[image]

**Expected behavior**

service vm should boot successfully.

**Is there something wrong with my way to boot ACRN?**

gvancuts, 2021-04-24:
Can you check/confirm the following:
* Number of CPUs on your platform?
* Check bios options (hyperthreading turned off, all VT turned on, e.g. VT-x, VT-d)
* Version of Ubuntu used, is that 18.04 or 20.04?

I don't know if this is what can cause this issue but you need to patch Grub in 18.04.

iaGuoZhi, 2021-04-24:
Thanks @gvancuts 
After I configure BIOS with hyperthreading disabled, ACRN can boot successfully now. Thanks a lot~

gvancuts, 2021-04-25:
Thanks for the confirmation @iaGuoZhi !

@NanlinXie @dbkinder , this is an example of an error message that could be improved. And perhaps ACRN shouldn't panic if there are more processors than what it was configured to use.

iaGuoZhi, 2021-04-26:
In [Getting Started Guide for ACRN Industry Scenario with Ubuntu Service VM](https://projectacrn.github.io/2.0/getting-started/rt_industry_ubuntu.html) with ACRN v2.0, Service VM should be installed in nvm while RTVM should be installed in SATA.

Is it necessary? 

I installed both Service VM and RTVM in SATA, while Service VM on `/dev/sda5 `and RTVM in `/dev/sda6`. However I came across such error when launch RTVM. I don't know if this has anything to do with installing Service VM and RTVM in the same device.

[image]

gvancuts, 2021-04-26:
Hi @iaGuoZhi , you will need to adjust the launch script and also make sure that you have a bootloader properly configured if you do that. Where did you put the ESP (Grub bootloader)?

Having said that, you will find that sharing the same physical disk (just different partition) will adversely affect the realtime performance of your RTVM. The reason we install the Service VM and RTVM rootfs on different partitions is so that we can pass-through the storage controller to the RTVM and we avoid sharing that resource.

iaGuoZhi, 2021-04-27:
Hi @gvancuts 

My ESP is in /dev/sda3.
[image]

and I modified RTVM grub file like this.
[image]
where UUID and PARTUUID and been updated according to RTVM's root partition `/dev/sda6`.

**I'm confused about how ACRN find RTVM's image?**

Here is my RTVM launch script which I commented disk pass-through.
[image]

fuzhongl, 2021-04-27:
@iaGuoZhi 
It is expected; setup related  issue: 
SATA was passthroughed to RTVM, but SOS also using this SATA.
We prefer RTVM installed in one Whole SSD(If no more SSD on your board, please using USB driver) then passthrough the whole SSD to RTVM, ovmf will launch it base on the bdf info in launch script.

gvancuts, 2021-04-27:
Hi @iaGuoZhi , I'm not sure what you call teh RTVM Grub config but in your case I suspect you have only one Grub instance for the entire system, and that's causing problems here. Here is what the Getting Started Guide is setting up when you use two separate disks in terms of boot sequences:
1. Service VM: platform UEFI firmware -> Grub (ESP partition on the **NVMe** disk) -> rootfs on NVMe (`/dev/nvme0p2`)
2. User VM (RTVM): virtual UEFI firmware (`OVMF.fd`) -> Grub (on the **SATA** disk) -> rootfs on SATA (`/dev/sda2`)

The key is that **each** VM (Service VM and RTVM) have their own Grub, and each is configured to pick up the right kernel and set the rootfs to the right disk partition.

There are a couple of options (well, maybe three actually) for you if you want to use a single disk:
1. Create a virtual disk image based on Ubuntu in which we will replace the stock kernel by a realtime kernel (that virtual disk will contain an ESP partition with its own Grub that we would configure for the RTVM, using the UUID and PARTUUID of the virtual disk)
2. Use a dedicated partition (like you are doing today):
   1. Use the `-k` and `-B` options from `acrn-dm` to use the RTVM kernel (that is located somewhere in your Service VM filesystem) and set the `bootargs` value to point at the partition where the RTMV rootfs is
   2. Treat the partition as an entire disk and install a copy of grub in there... it probably involves splitting that partition into at least two partitions, one to act as the ESP and the other to host the RTVM rootfs.

Note that the realtime performance in **any** of those scenarios is likely to be sub-optimal because you are sharing the same physical resource and the Service VM will therefore interfere with the RTVM (for storage access).

We do not have a good set of instructions for you for any of these set-ups as described above but if you pick one, we can guide you through it. My recommendation is that we try first with option 2.1 (using the `-k` and `-B` options of `acrn-dm` and point at the SATA partition for the RTVM rootfs).

We will also need to restore your Grub installation to point at the correct Service VM partition.

iaGuoZhi, 2021-04-27:
Thanks! @gvancuts @fuzhongl 
I'd like to try option2.1 first. I have two questions.
1. where should bootargs set? and which is bootargs's value? UID or bdf?
2.  _We will also need to restore your Grub installation to point at the correct Service VM partition._ Do you mean I should update file `/etc/grub.d/40_custom` of SOS?

iaGuoZhi, 2021-04-27:
Thanks! @gvancuts @fuzhongl
I'd like to try option2.1 first. I have two questions.

1. which is bootargs's value? UID or bdf?
2. _We will also need to restore your Grub installation to point at the correct Service VM partition._ Do you mean I should update file /etc/grub.d/40_custom of SOS?

gvancuts, 2021-04-27:
The `-B, --bootargs` parameter should be set to something like: `root=PARTUUID=<UUID of rootfs partition> rw rootwait nohpet console=hvc0 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll irqaffinity=0`

Re: Grub: yes, you need to make sure that the `/etc/grub.d/40_custom` file that's on the **Service VM partition (i.e. /dev/sda5?)** is updated to use the Service VM kernel parameters and correct PARTUUID (from `/dev/sda5`).

iaGuoZhi, 2021-04-27:
Thanks!, After follow above tips, I update acrn-dm command parameters, however my usage is not correct, here is my script.
[image]

iaGuoZhi, 2021-04-27:
> @iaGuoZhi
> It is expected; setup related issue:
> SATA was passthroughed to RTVM, but SOS also using this SATA.
> We prefer RTVM installed in one Whole SSD(If no more SSD on your board, please using USB [...]

Can I install RTVM in a usb flash disk, and pass through USB to RTVM?

iaGuoZhi, 2021-04-27:
Hi @gvancuts  I changed launch script like this:
[image]

However I came across same error as same as https://github.com/projectacrn/acrn-hypervisor/issues/5970#issuecomment-826776579

fuzhongl, 2021-04-27:
> > @iaGuoZhi
> > It is expected; setup related issue:
> > SATA was passthroughed to RTVM, but SOS also using this SATA.
> > We prefer RTVM installed in one Whole SSD(If no more SSD on your board, please using USB driver) then passthrough the whole SSD to RTVM, ovmf will launch it base on the bdf info in launch script.
> 
> Hi @fuzhongl Can I install RTVM in a usb flash disk, and pass through USB to RTVM?

@iaGuoZhi Yes. you can passthrough USB controller to RTVM.

taoyuhong, 2021-04-28:
About ‘Fix address space of post launched rfc v2 #5914’
My issue is OVMF can’t drive NVME ptdev at the pre-merge test env, but I can’t reproduce it on my whl-ipc-i7. 
If you use /user/share/acrn/bios/OVMFD_debug.fd, you may see more serial output, and find out the where is the problem.
By the way, have you tried virtio-blk? In my case, vm can launch with virtio-blk device, because it is nvme issue.

closed 2022-02-25 by fuzhongl