projectacrn/acrn-hypervisor #5972: configure one display port for Windows and another for SOS?

zhangshuai75, 2021-04-27:
Our paltform is MaxTang WL37, Intel(R) Core(TM) i7-8665U
It has two display ports, a VGA and a HDMI. When I launch the Windows UOS, both of them will show the Windows desktop.
If I want to operate on SOS again, I can only log in remotely through the network.

So, is it possible to configure one port for Windows and another for SOS？

gvancuts, 2021-04-27:
Hi @zhangshuai75 , you need to use GVT-g (graphics sharing) to be able to assign display ports to different VMs (here the Service VM and Windows User VM).

Unfortunately I do not think GVT-g still works in recent versions of ACRN, so you may have to go back to a previous version (from the 1.x series).

@NanlinXie @fuzhongl , can you help clarify the status of GVT-d in ACRN?

fuzhongl, 2021-04-27:
> Our paltform is MaxTang WL37, Intel(R) Core(TM) i7-8665U
> It has two display ports, a VGA and a HDMI. When I launch the Windows UOS, both of them will show the Windows desktop.
> If I want to operate [...]

@zhangshuai75 About both of them will show the Windows desktop, this is becaused gpu is passthroughed to Windows.
If you  want to configure one port for Windows and another for SOS; Please using v2.0 for gvt-g(since gvt-g has the known issue since v2.1)

zhangshuai75, 2021-04-27:
ok, I will try the v2.0 later, thanks.

closed 2021-04-27 by zhangshuai75