projectacrn/acrn-hypervisor #5984: GVT:init failed when run ubuntu as user vm in maxtang board

iaGuoZhi, 2021-05-06:
**Describe the bug**

After following toturial [Run Ubuntu as the User VM](https://projectacrn.github.io/2.4/tutorials/running_ubun_as_user_vm.html) to run ubuntu as user vm. I encountered an error.  The error message is GVT:init failed.

**Platform**
MaxTang whl-ipc-i5

**Codebase**
ACRN-Hypervisor: v2.4
ACRN-Kernel: v2.4
Service-vm: ubuntu20
User-vm: ubuntu18

**Scenario**
Industry

**To Reproduce**

Based on Maxtang whl-ipc-i5 board, follow toturial of run ubuntu as user vm, when exec command `sh ./launch_ubuntu.sh`, error message will show like this:
[image]
 

**Expected behavior**

Ubuntu user vm should boot successfully.

iaGuoZhi, 2021-05-06:
Does GVT enabled in ACRN v2.4 for WHL-IPC-I5?

fuzhongl, 2021-05-06:
@iaGuoZhi There is the known issue for gvt-g since v2.1.
Do you use gvt-g or gvt-d in your launch script?

iaGuoZhi, 2021-05-06:
Hi @fuzhongl 
I used param `-s 2,pci-gvt -G "$2" \` when exec acrn-dm command in launch_ubuntu.sh. This would call gvt-g? Then I encountered above error: `GVT: init failed`.

I also followed this document [Enable GVT-d in ACRN](https://projectacrn.github.io/latest/tutorials/gpu-passthru.html) to use GVT-d. changed launch_ubuntu.sh from `-s 2,pci-gvt -G "$2" \`  to `-s 2,passthru,0/2/0,gpu \` and enabled GVT-d GOP Driver. However ,After I run launch_ubuntu.sh, My monitor is not connected to my Board, and after about 5 seconds, My Board is rebooted. ubuntu user vm was never showed. This phenomenon is consistent with running windows as user vm in my whl-ipc-i5 board.

fuzhongl, 2021-05-07:
@iaGuoZ Do you follow the guide: https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#additional-settings-in-the-service-vm to set BIOS?

iaGuoZhi, 2021-05-07:
> @iaGuoZ Do you follow the guide: https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#additional-settings-in-the-service-vm to set BIOS?

This guide recommands using ubuntu 18.04 with grub version 2.04, However I failed to upgrade ubuntu18's grub from 2.02 to 2.04. Therefore I use ubuntu20 with grub 2.04 as service vm

iaGuoZhi, 2021-05-07:
> @iaGuoZ Do you follow the guide: https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#additional-settings-in-the-service-vm to set BIOS?
I tried to set BIOS according to `Recommended BIOS Setting for RTVM` in this guide. However BIOS settings depend on the platform. This guide is for WHL-IPC-I7, My board is WHL-IPC-I5. some bios setting it that table was not found in my bios. Would you mind share me BIOS Settings for WHL-IPC-I5?

fuzhongl, 2021-05-07:
@iaGuoZhi Your BIOS setting is set as following, right?
Set DVMT Pre-Allocated to 64MB
Set PM Support to Enabled

Please in input `vcpu_list` in ACRN console to check if UOS is launched.

iaGuoZhi, 2021-05-07:
> @iaGuoZhi Your BIOS setting is set as following, right?
> Set DVMT Pre-Allocated to 64MB
> Set PM Support to Enabled
> 
> Please in input `vcpu_list` in ACRN console to check if UOS is launched.

@fuzhongl DVMT Pre-Allocated is 64MB, PM Support is enabled. these bios requirements is satisfied.
Here are my console outputs:

#### GVT-d GPU passthrough

[image]

#### GVT-d GOP enabled 

[image]

zhangshuai75, 2021-05-10:
I have encountered windows uos restart problems before.
My solution is change the Ubuntu sos to text mode: enable “multi-user.target” with “sudo systemctl set-default multi-user.target ”

fuzhongl, 2022-03-22:
@zhangshuai75 Thanks for your input!
@iaGuoZhi Please have a try with:
```
“sudo systemctl set-default multi-user.target” then reboot.
or
Add i915.modeset=0 in Service VM kernel cmdline.
```

fuzhongl, 2022-07-05:
Close it since no update long time.

closed 2022-07-05 by fuzhongl