projectacrn/acrn-hypervisor #5992: Document question in gettting started guide for ACRN Hybrid mode

iaGuoZhi, 2021-05-08:
**Describe the bug**

Document error

**Codebase**

ACRN hypvisor 2.4 
ACRN kernel 2.4

**Scenario**

Hybrid

**To Reproduce**

There is no zephyr.bin or ACPI_VM0.bin in /boot directory.  See https://projectacrn.github.io/2.4/tutorials/using_hybrid_mode_on_nuc.html#id2

fuzhongl, 2021-05-08:
@iaGuoZhi You need to build them first; please refer the link to build Zephyr:
https://projectacrn.github.io/2.4/tutorials/using_zephyr_as_uos.html#introduction-to-zephyr
ACPI_VM0.bin will be generated for hybrid scenario: acrn-hypervisor/build/hypervisor/acpi/ACPI_VM0.bin

amyreye, 2022-02-17:
@gvancuts @dbkinder What is the status?

fuzhongl, 2022-03-22:
Closed as GSG is updated.

closed 2022-03-22 by fuzhongl