projectacrn/acrn-hypervisor #6003: QEMU: Failure in creating VMX controls : check_vmx_ctrl in hypervisor/arch/x86/guest/vmcs.c

kishorekanala, 2021-05-12:
**Describe the bug**
Failure during booting acrn.32.out (ACRN Hypervisor): VMX Control creation failing

**Platform**
QEMU 4.2.1 on HP Omen laptop
Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz

**Codebase**
ACRN Hypervisor 2.4

**Scenario**
Describe the scenario: Hybrid, Industry or logical_partition
Followed setup procedure in https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html

**To Reproduce**
Setup as given in https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html
Select ACRN Hypervisor in Grub

**Expected behavior**
ACRN Hypervisor should boot

**Additional context**
Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Multiboot Bootloader: GRUB 2.02-2ubuntu8.23
[0us][cpu=0][(null)][sev=3][seq=1]:vlapic: Basic apicv ops is used!
ACRN Hypervisor
calibrate_tsc, tsc_khz=2592235
[2084832us][cpu=0][(null)][sev=2][seq=2]:HV version 2.4-2021-05-11 18:07:18- DBG (daily tag:) sdc@qemu build by kishoreacrn, start time 2072994us
[2086640us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[2087323us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
ACRN:\>[2461933us][cpu=0][(null)][sev=2][seq=29]:Start VM id: 0 name: ACRN SOS VM
[2463120us][cpu=0][vm0:vcpu0][sev=3][seq=30]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[2464517us][cpu=0][vm0:vcpu0][sev=3][seq=31]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[3960186us][cpu=0][vm0:vcpu0][sev=3][seq=34]:vlapic: Start Secondary VCPU1 for VM[0]...
[3961622us][cpu=1][vm0:vcpu1][sev=3][seq=35]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[3963651us][cpu=1][vm0:vcpu1][sev=3][seq=36]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[4042481us][cpu=0][vm0:vcpu0][sev=3][seq=38]:vlapic: Start Secondary VCPU2 for VM[0]...
[4043858us][cpu=2][vm0:vcpu2][sev=3][seq=39]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[4045112us][cpu=2][vm0:vcpu2][sev=3][seq=40]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

[4121542us][cpu=0][vm0:vcpu0][sev=3][seq=42]:vlapic: Start Secondary VCPU3 for VM[0]...
[4123077us][cpu=3][vm0:vcpu3][sev=3][seq=43]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[4125074us][cpu=3][vm0:vcpu3][sev=3][seq=44]:VMX ctrl 0x48b not fully enabled: request 0x4ab but get 0xab

gvancuts, 2021-05-12:
Can you confirm if you have access to the ACRN shell at that point? (type Enter and you should see `ACRN:\>` if you do).

Assuming you do, check the Service VM console as well: `vm_console 0` and tell us what it says.

There is an existing issue against QEMU (see #5938), let's see if this is a duplicate.

kishorekanala, 2021-05-12:
Hi Geoffroy,

    Yes. ACRN shell is seen.
    Tried vm_console 0
    Ubuntu (Service VM) boots up and login prompt is seen.

>There is an existing issue against QEMU (see #5938
<https://github.com/projectacrn/acrn-hypervisor/issues/5938>), let's see if
this is a duplicate.
   Behaviour is slightly different for me. ACRN shell comes up for me and
service VM also boots up
   However, sometimes, I get the following error. When this happens, at
ACRN shell, vm_list doesn't list any VMs. [Hence, vm_console 0 says invalid
parameters

Loading ACRN hypervisor with SDC scenario ...
error: Can't get controller info..
Multiboot Bootloader: GRUB 2.02-2ubuntu8.23
[0us][cpu=0][(null)][sev=3][seq=1]:vlapic: Basic apicv ops is used!
ACRN Hypervisor
calibrate_tsc, tsc_khz=2591719
[2639579us][cpu=0][(null)][sev=2][seq=2]:HV version 2.4-2021-05-11
18:07:18- DBG (daily tag:) ***@***.*** build by kishoreacrn, start time
2628205us
[2640883us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[2641301us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel(R)
Core(TM) i7-9750H CPU @ 2.60GHz
ACRN:\>[2971433us][cpu=0][(null)][sev=2][seq=19]:Assertion failed in file
arch/x86/vtd.c,line 256 : fatal error
[2972631us][cpu=0][(null)][sev=2][seq=20]:
 delta = (actual_load_address - CONFIG_HV_RAM_START) = 0x0

[2973735us][cpu=0][(null)][sev=2][seq=21]:
Host Stack: CPU_ID = 0

[2974471us][cpu=0][(null)][sev=2][seq=22]:addr(0x1120aad0)
0x0000000000000000  0x00000000007d9bdc  0x000000001305d040
 0x00000001ca882b8c

[2975924us][cpu=0][(null)][sev=2][seq=23]:addr(0x1120aaf0)
0x000000001120ab30  0x000000001100c976  0x0000000011051160
 0x000000001305d040

[2977368us][cpu=0][(null)][sev=2][seq=24]:addr(0x1120ab10)
0x0000000000000000  0x000000001100c880  0x000000001103d134
 0x0000000000000000

[2978814us][cpu=0][(null)][sev=2][seq=25]:addr(0x1120ab30)
0x000000001120ab60  0x000000001100c362  0x000000001306d000
 0x000000001306d000

[2980256us][cpu=0][(null)][sev=2][seq=26]:addr(0x1120ab50)
0x0000000011051160  0x000000001306d000  0x000000001120abf0
 0x000000001101646e

[2981701us][cpu=0][(null)][sev=2][seq=27]:addr(0x1120ab70)
0x0000000000000000  0x0000000011051160  0x000000000000000f
 0x000000001120ac00

[2983148us][cpu=0][(null)][sev=2][seq=28]:addr(0x1120ab90)
0x000000000000000d  0x0000000000000000  0x000000017e300000
 0x0000000000000000

[2984589us][cpu=0][(null)][sev=2][seq=29]:addr(0x1120abb0)
0x00000000110450c7  0x839118c584437d48  0x0000000000000000
 0x0000000000000000

[2986034us][cpu=0][(null)][sev=2][seq=30]:addr(0x1120abd0)
0x0000000011051160  0x0000000000000000  0x0000000000000000
 0x0000000000000000

[2987358us][cpu=0][(null)][sev=2][seq=31]:addr(0x1120abf0)
0x000000001120ac30  0x000000001101701c  0x0000000000000000
 0x839118c584437d48

[2988404us][cpu=0][(null)][sev=2][seq=32]:addr(0x1120ac10)
0x000000001120ac50  0x0000000000000000  0x0000000000000000
 0x0000000000000000

[2989448us][cpu=0][(null)][sev=2][seq=33]:addr(0x1120ac30)
0x000000001120ac50  0x0000000011017128  0x0000000000000000
 0x0000000000000000

[2990490us][cpu=0][(null)][sev=2][seq=34]:addr(0x1120ac50)
0x000000001120ac70  0x00000000110081e8  0x00000000110010e2
 0x0000000000000000

[2991538us][cpu=0][(null)][sev=2][seq=35]:addr(0x1120ac70)
0x00000000110548b0  0x00000000696e746c  0x0000000000000000
 0x0000000000000000

[2992583us][cpu=0][(null)][sev=2][seq=36]:addr(0x1120ac90)
0x0000000100000000  0x0000000000000000  0x0000000000000000
 0x0000000000000000

[2993622us][cpu=0][(null)][sev=2][seq=37]:addr(0x1120acb0)
0x000000001120acb0  0x000000001120acb0  0x0000000000000000
 0x0000000000000000

[2994663us][cpu=0][(null)][sev=2][seq=38]:

[2995020us][cpu=0][(null)][sev=2][seq=39]:Host Call Trace:

[2995677us][cpu=0][(null)][sev=2][seq=40]:----> 0x000000001100c976

[2996422us][cpu=0][(null)][sev=2][seq=41]:----> 0x000000001100c362

[2997173us][cpu=0][(null)][sev=2][seq=42]:----> 0x000000001101646e

[2997922us][cpu=0][(null)][sev=2][seq=43]:----> 0x000000001101701c

[2998665us][cpu=0][(null)][sev=2][seq=44]:----> 0x0000000011017128

[2999423us][cpu=0][(null)][sev=2][seq=45]:----> 0x00000000110081e8

[3000164us][cpu=0][(null)][sev=2][seq=46]:

Regards,
Kishore

gvancuts, 2021-05-12:
Thanks @kishorekanala . I have increased the loglevel output for the hypervisor on my system and now I am seeing the same issue than you - see https://github.com/projectacrn/acrn-hypervisor/issues/5938#issuecomment-839633877.

What is your loglevel?

kishorekanala, 2021-05-12:
ACRN:\>loglevel
console_loglevel: 3, mem_loglevel: 5, npk_loglevel: 5

gvancuts, 2021-05-12:
> 
> 
> ACRN:>loglevel
> console_loglevel: 3, mem_loglevel: 5, npk_loglevel: 5

Thanks. I find it interesting to see that the loglevel value does seem to affect what happens on my system. If I increase it to 5 or 6, I get this error consistently (at least so far, and I have tried a dozen times).

Can you try to increase that value to `5` in your code and see if that makes it more consistently happening for you too?

kishorekanala, 2021-05-12:
I modified config.h and built acrn.32.out.
I am stuck with https://github.com/projectacrn/acrn-hypervisor/issues/6005 problem. Nothing to do with my modifications. This used to happen earlier also. Just that now it is consistent. Not able to succeed.

kishorekanala, 2021-05-12:
When I set to log level to 5, behavior is same. So, loglevel does seem to affect the behavior.
ACRN:\>loglevel
console_loglevel: 5, mem_loglevel: 5, npk_loglevel: 5
ACRN:\>vm_list

VM_UUID                          VM_ID VM_NAME                          VM_STATE
================================ ===== ================================ ========
dbbbd4347a574216a12c2201f1ab0240   0   ACRN SOS VM                      Running

fuzhongl, 2021-05-14:
@kishorekanala 
v2.0 is the preferred  version for ACRN on qemu; Do you see the same issue with V2.0?

gvancuts, 2021-06-02:
@kishorekanala , we are in the final stages of publishing an updated tutorial for QEMU based on ACRN v2.4 - see #5937. v2.5 is blocked by #6102

fuzhongl, 2021-06-04:
@kishorekanala 
Will close this issue since [#6102](https://github.com/projectacrn/acrn-hypervisor/issues/6102) is fixed in v2.5; please have a try.
Any question, please re-open this issue.
Thanks!

closed 2021-06-04 by fuzhongl

gvancuts, 2021-06-04:
Note that there is still #6132 that is under investigation. It affects ACRN on QEMU but it is a transient issue (one you also reported earlier @kishorekanala : `Assertion failed in file arch/x86/vtd.c,line 256 : fatal error`.

fuzhongl, 2021-06-11:
@gvancuts and @kishorekanala 
`Assertion failed in file arch/x86/vtd.c,line 256 : fatal error` is fixed with following patch, will PR and merge it later.
```
--- a/hypervisor/arch/x86/vtd.c
+++ b/hypervisor/arch/x86/vtd.c
@@ -252,8 +252,13 @@ static inline void dmar_wait_completion(const struct dmar_drhd_rt *dmar_unit, ui

        do {
                *status = iommu_read32(dmar_unit, offset);
-               ASSERT(((rdtsc() - start) < CYCLES_PER_MS),
-                       "DMAR OP Timeout!");
+               if ((rdtsc() - start) > CYCLES_PER_MS) {
+                       pr_err("%s %d DMAR OP Timeout", __func__, __LINE__);
+                       start = rdtsc();
+               }
+
+       //      ASSERT(((rdtsc() - start) < CYCLES_PER_MS),
+       //              "DMAR OP Timeout!");
                asm_pause();
        } while( (*status & mask) == pre_condition); 
```