projectacrn/acrn-hypervisor #6005: File size of acrn.32.out becoming zero

kishorekanala, 2021-05-12:
**Describe the bug**
After copying acrn.32.out to /boot, when I start ACRNSOS and choose ACRN hypervisor in grub and try to boot hypervisor, grub says acrn.32.out is not found. The file size of acrn.32.out is becoming zero.
I made it read-only for all users. Even then, it doesn't help. This happens only sometimes. Sometimes consistent. I need to keep copying the out file in /boot/ file and keep trying.

**Platform**
QEMU 4.2.1 on HP Omen (Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz)

**Codebase**
2.4 version of Hypervisor

**Scenario**
Describe the scenario: Hybrid, Industry or logical_partition

**To Reproduce**
Build hypervisor
Copy acrn.32.out to /boot/
virsh destroy ACRNSOS
virsh start ACRNSOS
select ACRN Hypervisor
It says acrn.32.out file not found

**Expected behavior**
ACRN hypervisor should boot

**Additional context**
oading ACRN hypervisor with SDC scenario ...
error: file `/boot/acrn.32.out' not found.
error: you need to load the kernel first.

Press any key to continue...

gvancuts, 2021-05-12:
Wow... that I have never seen. The steps you describe to reproduce are the ones I use all the time to test ACRN on QEMU. Well, I use the `--console` parameter when calling `virsh start` but I guess you do too?

My host is Ubuntu 20.04 and the QEMU version I run is the same as yours.

When that happens, are you still able to boot the stock Ubuntu image with no apparent issue?

Is there anything on your set-up that may be specific compared to the most traditional installation (disk images hosted on remote filesystem, etc.)? I know this is a very open-ended question but I'm trying to find clues as to what is going on.

kishorekanala, 2021-05-12:
Correct. Steps are the same.
My host is also Ubuntu 20.04, QEMU is also 4.2.1. True.. it is same as yours.
When I press any key, it goes back to Grub menu, I select GNU/Linux. Service OS comes up without any problem, I ssh to Service OS and continue to copy the acrn.32.out to /boot/ and keep retrying till it succeeds
Setup :  Nothing specific.

fuzhongl, 2021-05-14:
@kishorekanala 
We don't met this issue for v2.0 so far. 
v2.0 is the preferred version for ACRN on qemu; please try v2.0 if there is the same issue on your side.
If yes, please share the reproduce rate.
Thanks!

fuzhongl, 2021-05-14:
> Correct. Steps are the same.
> My host is also Ubuntu 20.04, QEMU is also 4.2.1. True.. it is same as yours.
> When I press any key, it goes back to Grub menu, I select GNU/Linux. Service OS comes up [...]

@kishorekanala Do you type `sync` after copy the acrn.32.out to /boot/ ?

fuzhongl, 2022-03-22:
Should be setup issue since it can't be reproduced on our side.

closed 2022-03-22 by fuzhongl