projectacrn/acrn-hypervisor #6033: Fix HV page fault when shutting down post-launched RTVM

taoyuhong, 2021-05-19:
**Describe the bug**
A clear and concise description of what the bug is.

**Platform**
Describe the platform information, 

**Codebase**
Describe which codebase is used.

**Scenario**
Describe the scenario: Hybrid, Industry or logical_partition

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Attach the following logs if available:

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

taoyuhong, 2021-05-19:
[External_System_ID] ACRN-7140

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie