projectacrn/acrn-hypervisor #6092: SOS does not boot on custom platform

villanyibalint, 2021-05-26:
**Describe the bug**
The Acrn SOS does not boot. The SOS VM0 hangs on startup.

**Platform**
Custom platform based on Intel Xeon Silver 4109T

**Codebase**
HV: HV 2.5-unstable-2021-05-25 [acrn-2021w21.4-180000p](https://github.com/projectacrn/acrn-hypervisor/releases/tag/acrn-2021w21.4-180000p)
SOS: Debian 10 with [Acrn kernel 2.4](https://github.com/projectacrn/acrn-kernel/releases/tag/v2.4)

**Scenario**
Custom scenario posted below. I just want to test the hypervisor, the scenario includes the service OS and an RT post-launch VM.

**To Reproduce**
1. Followed tutorial [Run Debian as the Service VM](https://projectacrn.github.io/latest/tutorials/running_deb_as_serv_vm.html?highlight=debian)
2. On startup the hypervisor initializes, but the SOS never boots.

**Expected behavior**
The SOS boots successfully.

**Additional context**
I understand that I am running an unstable build, the reason is that the platform has an address width of 46, which is not supported in 2.4 yet. I also understand that as a custom platform, you might not be able to fully assist me.

This is the visual output from the machine:
[image]

This is the ACRN serial log:
[log_700102-000613.log](https://github.com/projectacrn/acrn-hypervisor/files/6547245/log_700102-000613.log)

This is my scenario configuration:
[customscenario.txt](https://github.com/projectacrn/acrn-hypervisor/files/6547352/customscenario.txt)

This is what I could deduce from the ACRN log:
* the last VM exit is a halt instruction, nothing happens afterwards
* some of the sequence numbers in the logs are inconsistent, some numbers are missing. I am not sure if this is expected behaviour, but possibly some messages are missing from the end of the log as well.
* this is what happens when I try to enter the VM shell from ACRN:
[image]
After this, nothing happens.

My main issue with solving the problem on my own has been that I cannot see anything that happens inside the VM. I have no output from the SOS, neither on the visual output, the ACRN logs, or the serial port.

Any help is greatly appreciated.

gvancuts, 2021-05-26:
Hi @villanyibalint , one thing you could try is to patch the ACRN source code with this https://github.com/tttech-industrial-buchsbaum/acrn-hypervisor/blob/debian/master/debian/patches/0008-hv-add-vm_console-command-line-parameter.patch

Rebuild it and also add the `vm_console=0` parameter in your Grub config (Example: `multiboot2 /boot/acrn/acrn.bin  root=PARTUUID="e515916d-aac4-4439-aaa0-33231a9f4d83" vm_console=0`)

This will make ACRN switch to the SOS console as soon as it starts. It will be interesting to see if anything gets printed out, because what could happen is that the Service VM (SOS) boot process is dead by the time you manually type and switch to that console.

Have you ever tried using Ubuntu as the Service VM OS, and if so what were the results?

Are you also using a customer BOARD xml when building ACRN?

villanyibalint, 2021-05-26:
Hi, thanks for the suggestion that looks like exactly what I need.

I have not tried Ubuntu yet, but if this doesn't work out I'll give it a shot. And yes I am also using a custom board XML generated with the config tool. If it's not absolutely necessary I would hold on with posting that because of confidentiality, but if the SOS logging does not work out I'll go clean it up and post it as well.

gvancuts, 2021-05-26:
> I have not tried Ubuntu yet, but if this doesn't work out I'll give it a shot.

Sounds good!

> And yes I am also using a custom board XML generated with the config tool. If it's not absolutely necessary I would hold on with posting that because of confidentiality, but if the SOS logging does not work out I'll go clean it up and post it as well.

No need at the moment. I was mostly (so far) interested to check the memory configuration, to make sure it matches what you have on your custom board. From your scenario file:
```
    <MEMORY>
      <STACK_SIZE>0x2000</STACK_SIZE>
      <HV_RAM_SIZE></HV_RAM_SIZE>
      <HV_RAM_START></HV_RAM_START>
      <LOW_RAM_SIZE>0x00010000</LOW_RAM_SIZE>
      <PLATFORM_RAM_SIZE>0x400000000</PLATFORM_RAM_SIZE>
    </MEMORY>
```

villanyibalint, 2021-05-26:
The SOS shell seems to be empty from the start:
[log_700101-034446.log](https://github.com/projectacrn/acrn-hypervisor/files/6548232/log_700101-034446.log)

I can see the message pop up "Entering VM 0 shell" but nothing after.

Can you advise me on the memory configuration? This is the memory section part of the board config:
```xml
<memory>
  <range start="0x0000000000000000" end="0x000000000006efff" size="454656"/>
  <range start="0x0000000000070000" end="0x000000000007ffff" size="65536"/>
  <range start="0x0000000000100000" end="0x0000000086dfefff" size="2261774336"/>
  <range start="0x0000000093fff000" end="0x0000000093ffffff" size="4096"/>
  <range start="0x0000000100000000" end="0x0000003057ffffff" size="203339857920"/>
</memory>
```
The board itself has 192GiB memory I think but I saw somewhere that Acrn only supports up to 16GiB so I just left the capacity as is. Here is the complete board config as well: [board.txt](https://github.com/projectacrn/acrn-hypervisor/files/6548496/board.txt)

fuzhongl, 2021-05-27:
@villanyibalint 
Could you please add `"nokaslr"` in sos kernel cmdline?
Adding it at the end of multiboot2; just like following:

`multiboot2 /boot/acrn/acrn.bin  root=PARTUUID="e515916d-aac4-4439-aaa0-33231a9f4d83"  nokalsr`

closed 2021-05-27 by villanyibalint

fuzhongl, 2021-05-27:
@villanyibalint
Thanks for confirm!
Adding "nokaslr" in sos kernel cmdline is work around; we will fix this memory related issue later.