projectacrn/acrn-hypervisor #6102: Build failure for BOARD=qemu SCENARIO=sdc on `release_2.5`

gvancuts, 2021-05-27:
It is not possible to build ACRN for QEMU on the `release_2.5` branch. The build fails as follows:
```
$ make BOARD=qemu SCENARIO=sdc
make[1]: Entering directory '/home/acrn/acrn-hypervisor/hypervisor'
Board XML is fetched from /home/acrn/acrn-hypervisor/misc/config_tools/data/qemu/qemu.xml
Scneario XML is configuration fetched from /home/acrn/acrn-hypervisor/misc/config_tools/data/qemu/sdc.xml
Traceback (most recent call last):
  File "/home/acrn/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 41, in <module>
    main(args)
  File "/home/acrn/acrn-hypervisor/misc/config_tools/static_allocators/main.py", line 31, in main
    module.fn(board_etree, scenario_etree, allocation_etree)
  File "/home/acrn/acrn-hypervisor/misc/config_tools/static_allocators/gpa.py", line 305, in fn
    create_native_pci_hole_node(allocation_etree, native_low_mem, native_high_mem)
  File "/home/acrn/acrn-hypervisor/misc/config_tools/static_allocators/gpa.py", line 238, in create_native_pci_hole_node
    common.append_node("/acrn-config/hv/MMIO/MMIO32_START", hex(low_mem[0].start).upper(), allocation_etree)
IndexError: list index out of range
make[1]: *** No rule to make target '/home/acrn/acrn-hypervisor/build/hypervisor/configs/allocation.xml', needed by '/home/acrn/acrn-hypervisor/build/hypervisor/configs/unified.xml'.  Stop.
make[1]: Leaving directory '/home/acrn/acrn-hypervisor/hypervisor'
Makefile:106: recipe for target 'hvdefconfig' failed
make: *** [hvdefconfig] Error 2
```
ACRN version:
```
$ git describe
acrn-2021w22.3-180000p-11-g701f6bf05
```

gvancuts, 2021-05-27:
@junjiemao1 I suspect this may have to do with recent ACRN Configuration Tools changes. Can you (or someone else) take a look? Thanks!

junjiemao1, 2021-05-31:
@NanlinXie It seems the board XML for qemu is not yet updated. Do we have any plan to do this?

gvancuts, 2021-05-31:
> @NanlinXie It seems the board XML for qemu is not yet updated. Do we have any plan to do this?

How do you create such new board XML?

yuchuyang, 2021-06-03:
[External_System_ID] ACRN-7190

gvancuts, 2021-06-03:
Patch was merged... closing!

closed 2021-06-03 by gvancuts