projectacrn/acrn-hypervisor #6108: Ubuntu service vm运行RTVM异常

cuipeng1993, 2021-05-28:
描述错误有关错误
运行RTVM系统出错：运行结果如下：
系统卡到这个地方：
 add_cpu
                                     out instr on NMI port (0x61) not supported
                                                                               out instr on NMI port (0x61) not supported
                                                                                                                        out instr on NMI port (0x61) not supported

平台
Phoenix Hill Platform with Intel®
Whiskey Lake U (cpu i5)

代码
ACRN Hypervisor V2.4
acrn-kernel v2.4
RTVM acrn-kernel 4.19/preempt-rt

方案
industry

重现
我只配置了launch_hard_rt_vm-bak.sh中sata相关参数
运行该脚本，有以下输出
root@sos:/tools# ./launch_hard_rt_vm-bak.sh
cpu1 online=1
cpu2 online=1
cpu3 online=1
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: hard_rtvm
VHM api version 1.0
vm_setup_memory: size=0x40000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/hard_rtvm/495AE2E526034D64AF76D4BC5A8EC0E5
open hugetlbfs file /run/hugepage/acrn/huge_lv2/hard_rtvm/495AE2E526034D64AF76D4BC5A8EC0E5
level 0 free/need pages:0/1 page size:0x200000
level 1 free/need pages:12/1 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 1 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x40000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7f5241bda000 -> baseaddr 0x0x7f5280000000
mmap 0x40000000@0x7f5280000000
touch 1 pages with pagesz 0x40000000
mmap 0x200000@0x7f537fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x40000000, biosmem 0x0, highmem 0x0
vm_init_vdevs
No pm notify channel given
polling 34...
Listening 34...
pci init hostbridge
pci init passthru
passthru device 0/17/0 BAR 1: size[100] is expanded to page aligned [1000]
passthru device 0/17/0 BAR 5: size[800] is expanded to page aligned [1000]
pci init virtio-console
tpm: init_vtpm2:Invalid socket path!
                                    acrn_sw_load
                                                SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 0x200000 copied to addr 0x7f537fe00000
                       SW_LOAD: build e820 4 entries to addr: 0x7f52800ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x0000000000100000, size 0x000000003ff00000,  type 0x1
SW_LOAD: entry[2]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[3]: addr 0x0000000140000000, size 0x0000000000000000,  type 0x2
[... 906 lines cut ...]
问题发生时的服务VM内核消息
root@sos:/home/sosacrn# dmesg -c
[  108.253571] smpboot: CPU 1 is now offline
[  108.254918] vhm: try to offline cpu 1 with lapicid 2
[  108.277660] smpboot: CPU 2 is now offline
[  108.280606] vhm: try to offline cpu 2 with lapicid 4
[  108.307440] NOHZ: local_softirq_pending 80
[  108.312340] smpboot: CPU 3 is now offline
[  108.318354] vhm: try to offline cpu 3 with lapicid 6
[  108.328295] ata1.00: disabled
[  108.334053] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[  108.334095] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[  108.334097] sd 0:0:0:0: [sda] Stopping disk
[  108.334106] sd 0:0:0:0: [sda] Start/Stop Unit failed: Result: hostbyte=DID_BAD_TARGET driverbyte=DRIVER_OK
[  108.350395] pci-stub 0000:00:17.0: claimed by stub
[  108.406526] vhm_dev_open: opening device node
[  108.409077] vhm-ioreq: init request buffer @ 000000002f7edb9b!
[  108.409084] vhm-ioreq: created ioreq client 1 for ioeventfd-2
[  108.412889] ACRN vhm ioeventfd init done!
[  108.412900] ACRN vhm irqfd init done!
[  108.412902] vhm: VM 2 created
[  108.962221] vhm-ioreq: created ioreq client 2 for acrndm

acrn-dm日志：/ var / log / acrn-tm中的所有文件(无acrn-tm文件)
运行脚本以启动来宾VM时的控制台日志(如何操作)

fuzhongl, 2021-05-28:
@cuipeng1993 Please share your launch script.

cuipeng1993, 2021-05-29:
我的附件中有啊，我已经上传了，你再看下第一封附件
---原始邮件---
发件人:&quot;Fuzhong ***@***.***&gt;
发送时间:2021年5月28日 星期五 下午8:20
***@***.***&gt;
***@***.******@***.***&gt;
主题:Re: [projectacrn/acrn-hypervisor] Ubuntu service vm运行RTVM异常 (#6108)

@cuipeng1993Please&nbsp;share&nbsp;your&nbsp;launch&nbsp;script.

&mdash;
You&nbsp;are&nbsp;receiving&nbsp;this&nbsp;because&nbsp;you&nbsp;were&nbsp;mentioned.
Reply&nbsp;to&nbsp;this&nbsp;email&nbsp;directly,view&nbsp;it&nbsp;on&nbsp;GitHub,&nbsp;orunsubscribe.

fuzhongl, 2021-05-30:
@cuipeng1993 Only log found; no launch script.
Please upload it.
Thanks!

fuzhongl, 2022-03-07:
Close it since no feedback long time.

closed 2022-03-07 by fuzhongl