projectacrn/acrn-hypervisor #6132: array-index-out-of-bounds in arch/x86/vtd.c

yonghuah, 2021-06-02:
Malicious input may trigger hypervisor buffer overflow in /arch/x86/vtd.c
*__________________________*
{noformat}
[121968253us][cpu=0][vm0:vcpu0][sev=3][seq=34]:UBSAN: array-index-out-of-bounds in arch/x86/vtd.c:1299:42
 [121968678us][cpu=0][vm0:vcpu0][sev=3][seq=35]:index 1023 is out of range for type 'uint64_t [4]'
[121969079us][cpu=0][vm0:vcpu0][sev=3][seq=36]:UBSAN: array-index-out-of-bounds in arch/x86/vtd.c:1416:68
 [121969499us][cpu=0][vm0:vcpu0][sev=3][seq=37]:index 1023 is out of range for type 'uint64_t [4]'
{noformat}

yonghuah, 2021-06-02:
[External_System_ID] ACRN-7102

gvancuts, 2021-06-03:
Is this related/the same as this error: `Assertion failed in file arch/x86/vtd.c,line 256 : fatal error`?

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie