projectacrn/acrn-hypervisor #6145: Add support to assign 3rd non-contiguous HPA regions for logical partition scenario

donsheng, 2021-06-03:
**Description**
Currently acrn supports 2 non-contiguous HPA regions (HPA1 and HPA2), extend
the code to support 3rd non-contiguous HPA region (HPA3) for logical partition scenario
(hybrid scenario is not supported).

To keep things simple, current design has the following assumptions for
ve820 and ept mapping:
1. HPA2 will always be placed after HPA1
2. HPA3 will always be placed after HPA2
3. HPA1/HPA2/HPA3 don’t share a single ve820 entry.
(Create multiple entries if needed but not shared)

**Platform**
Andersonlake

**Codebase**
V2.3

**Scenario**
Logical_partition

**To Reproduce**
they discovered that ACRN 2.3 assigns only the lower ASIL region and green to the DMA driver

**Expected behavior**
Both the lower ASIL region and higher ASIL region should be assigned to guest

donsheng, 2021-06-03:
[External_System_ID] ACRN-7192

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie