projectacrn/acrn-hypervisor #6202: virtio_console use-after-free

shuox, 2021-06-11:
ERROR: AddressSanitizer: heap-use-after-free on address 0x621000008680 at pc 0x55adb6cb87b4 bp 0x7fffac364020 sp 0x7fffac364010
 READ of size 4 at 0x621000008680 thread T0 (mevent)
 #0 0x55adb6cb87b3 in virtio_console_close_all hw/pci/virtio/virtio_console.c:1064
 #1 0x55adb6cb8884 in virtio_console_deinit hw/pci/virtio/virtio_console.c:1176
 #2 0x55adb6caf7a5 in pci_emul_deinit hw/pci/core.c:1048
 #3 0x55adb6cb0241 in deinit_pci hw/pci/core.c:1620
 #4 0x55adb6da655d in vm_deinit_vdevs core/main.c:535
 #5 0x55adb6daa2f2 in main core/main.c:1101
 #6 0x7f98f446f09a in __libc_start_main ../csu/libc-start.c:308
 #7 0x55adb6a80dfd in _start (/usr/bin/acrn-dm+0x1eadfd)

shuox, 2021-06-11:
[External_System_ID] ACRN-7103

NanlinXie, 2022-03-24:
Closed since patch merged without regression.

closed 2022-03-24 by NanlinXie