projectacrn/acrn-hypervisor #6206: CPU affinity and sharing documentation to be updated

gvancuts, 2021-06-14:
ACRN has introduced the ability to share CPU cores across multiple VMs (CPU sharing). This is documented in https://projectacrn.github.io/latest/tutorials/cpu_sharing.html. There are a couple of places in our documentation that we need to update regarding this feature:
* https://projectacrn.github.io/latest/tutorials/cpu_sharing.html: this document talks about `Kconfig` still. We need to update it to reflect the latest configuration option and tools.
* The "ACRN High-Level Design Overview" still mentions this feature as 'ToDO" - see https://projectacrn.github.io/latest/developer-guides/hld/hld-overview.html?highlight=todo#freedom-from-interference
* https://projectacrn.github.io/latest/developer-guides/hld/hv-cpu-virt.html also likely need to be updated to reflect the latest feature set

dbkinder, 2022-01-26:
@gvancuts I think we've addressed most of this, do you?

gvancuts, 2022-01-27:
> @gvancuts I think we've addressed most of this, do you?

I would agree, although we'll have to update many of these again now that the `--cpu_affinity` parameter definition is changing.